prusa3d/PrusaSlicer #3057: [Feature Request] Custom Pause G-Code

sarf2k4, 2019-10-11:
I noticed that on the recent stable prusa slicer release, there is pause feature implemented at certain layer height. The problem with this is usually works with most single material type printers while this pause feature may not work for other type of printer's firmware that uses different gcode pause command. I believe it is M600 for pause? basically it is hardcoded in prusa slicer.

I would like to request a custom field for pause sequence under printer specific g-code section. With this, printer owners especially prusa mk3s mmu2s, can switch between f1-f5 by using these kind of custom gcode.

These custom pause g-code sequence will be injected automatically when the pause feature at specific layer added. If the pause custom g-code is empty, the defailt M600 should be issued instead rather than not being inserted.

For example, with custom pause gcode sequence for mmu2 owner
```
1. Move the extruder aside via relative
2. Do filament blob shaping sequence
3. Move mmu2 head to the next/chosen filament 1-5
4. Purge the remnants of the previous filament
5. Move the extruder back via relative
6. Instruct printer to use the respective movement mode; absolute/relative
```

With the above example, I also thought of another good feature to integrate with this. Since mk3s+mmu2s understands T[toolhead] command, there should be a dropdown next to the '+' which is actually the filament/toolhead to switch to. This applies to any printers that has multiple extruders including mmu2 which has single nozzle, multiple extruder modes. With this, another new usable variable will be introduced that can be used maybe either only for pause gcode section or universally "[pause_toolhead]" maybe? Picture below for reference what I meant adding dropdown

[image]

I think, this is what has been missing that any mmu owners been waiting for and they can tweak to themselves depending on their needs without like the mmu1 prusa slicer days, everything hardcoded.  I hope this will be implemented and layer colour change can be automated especially those who owns mmu.

mroek, 2019-10-11:
Having a hardcoded pause command is a bad idea, since it makes the feature unuseable for printers that needs a different command to pause, so I think this request is a good idea.

sarf2k4, 2019-10-12:
> 
> 
> Having a hardcoded pause command is a bad idea, since it makes the feature unuseable for printers that needs a different command to pause, so I think this request is a good idea.

I don't think there's any slicer that features custom pause gcode commands yet. I know it is just  as simple as M600 issued and the printer will take care of the rest, but that M600 printer feature is  tedious if you're troubleshooting it which always have to reflash the firmware. Very handy for any multi material/extruder printers

mroek, 2019-10-12:
> I don't think there's any slicer that features custom pause gcode commands yet. I know it is just as simple as M600 issued and the printer will take care of the rest, but that M600 printer feature [...]

The usual method is to use scripting, and that is of course possible also in PrusaSlicer. The point here is that the builtin pause feature in PS is easier to use, it just needs customizable g-code (and more than one line as well) to be inserted for each pause location instead of the hardcoded M600.

It is of course possible to send the generated g-code though a separate post-processor (sed, sfk, perl, python) to replace any M600 with whatever you want, but that's an added inconvenience.

lukasmatena, 2019-10-12:
This is currently being implemented by @YuSanka. Her intention is to add a context menu to allow insertion of either M600, pause gcode, custom gcode or a forced toolchange (with multimaterial printers). This will be part of PrusaSlicer2.2.0.

I'm closing this as a duplicate of #2544, #1656, #1725, #2351, #1899 and maybe others.

closed 2019-10-12 by lukasmatena

supermerill, 2019-11-26:
@lukasmatena How this new feature that is planned for 2.2.0 is different from the "tool-change gcode" field?

lukasmatena, 2019-11-27:
The custom gcode can contain anything, not just a toolchange. Toolchange gcode is only inserted where a tool is changed and only for MM printers. But you are right if you're saying that most of it was already possible to do by scripting in custom gcodes (mostly after layer gcode). This will just be much more convenient and will not require messing with Printer Settings.