prusa3d/PrusaSlicer #3058: Thin wall detection causes miniature loops to be added, and emits empty G1 commands

mroek, 2019-10-11:
I know that there are several issues already regarding thin walls, and this one may well be the same as one of the other (at least as to the cause), but I think it still warrants a separate report.

Slicing the attached model with the attached config (latest 2.1 release of PS) gives this result:

[image]

As you can see, in the middle of a single strand gap fill there are multiple tiny loops generated, which is not the expected result. In this case it doesn't really do much harm, but it is clearly not correct.

In the g-code file PrusaSlicer also emits a lot of empty G1 commands in this area (for the attached model only layer 18 shows these issues). If thin wall detection is turned off, then these small loops disappear, but instead there are multiple added moves that appear as gaps (in the preview) further down in the model:

[image]

In other words, issues whether thin wall detection is off or on, just different ones.

[thin_wall_issue.zip](https://github.com/prusa3d/PrusaSlicer/files/3719514/thin_wall_issue.zip)

Jebtrix, 2019-10-11:
Related to #3025, #2666
Tweaking extrusion width (slightly smaller) can nudge the math here to help with this bug. Scaling object +-1~2% can also help if accuracy you need is flexible. Since you didn't attach a 3MF project file I don't know what print settings you used in that picture. You could also reduce number of perimeters either globally or with modifiers as a workaround when an odd number of perimeters is trying to fill entire circular feature.

mroek, 2019-10-11:
Yes, trial and error with settings can be done to work around the issue, but the core bug still needs to be fixed, I think.

The print settings (and model) was attached in the zip file, btw.

Jebtrix, 2019-10-11:
Bug is known and needs to be fixed absolutely. Yes you attached settings bundle and stl. That doesn't tell me which one of the print profiles you used.

mroek, 2019-10-11:
Ah. I'm new to PS, and didn't realize all settings would be exported, and not the active one. I also tried .3mf, but I couldn't see that the settings actually was included in that file when I loaded it back into PS.

Jebtrix, 2019-10-11:
No problem :) 3MF project is the best way for testing issues. It will be least amount of work for those reviewing it. Rest assured your current objects, print, printer, and filament profile is embedded. You can unzip the 3MF project file (might need to add .zip extension) and you can view the files to see your profiles inside ie Metadata\Slic3r_PE.config

mroek, 2019-10-11:
Thanks, that is good to know!

Jebtrix, 2019-10-11:
What might be confusing is how the profile names look when you load a 3MF file. Once their embedded think of them as profiles of the project now. They become disconnected is best way to describe it. If you wanted to use an embedded profile for other projects you need to Save a copy first. One of the biggest downsides to 3MF files currently is your parts become disconnected too, so Reload from Disk no longer works.

bubnikv, 2021-11-02:
The empty G1 lines are produced by the cooling logic. Fixed with 4e9f906848f6bbe31476e355919692ec1fb7b40d

    Cooling buffer used to remove F word from G1 Fxxx lines if the feed rate
    did not change compared to the preceding value creating empty "G1" lines.
    The empty "G1" lines are now removed.

closed 2026-09-29 by prusaslicer-bot[bot]