prusa3d/PrusaSlicer #3059: xy compensation

cdedwards, 2019-10-11:
Just seen a video that explained filament shrinkage and went looking for a method to increase/decrease XY dimensions according to filament selection. I found a XY compensation for this, but I believe it's really in the wrong location. This should be a filament type selection and not a print settings selection. Every filament type (PLA/PETG/ABS etc) has a differant shrinkage rate. So setting the shrinkage rate in the print settings tab is not really a great place. What's good for ABS is not good for PLA etc.

haarp, 2019-10-15:
Another reason for #3067 :)

rennne, 2019-11-21:
For filaments it should be in percent instead of mm. Since for bigger objects it is more distance but the same percentage.

6ffm70, 2020-07-31:
This is also, what I'm looking for: the scale feature for the plater is one thing; though scaling to provision for material shrinkage should be an option in the `Filament Settings`, as the shrinkage is not only individual to the plastic type, but even for different manufacturers and colours.

For every new material, we print a shrinkage sample (20mm³ w/ 1,2,3,4mm wall thinkness), and measure it in all three directions outside, and in two directions inside,
[image]
then we calculate the deviations into %, and store the values elsewhere, only to enter them manually in plater scaling every time.

This could be easier by having them available in the filament settings, and applied before slicing, as they are roughly always them same for each material/colour.

Here's -even if for a more distant future- #3687 an idea on how to make the provision values even more reproducible and reliable,

pipe2null, 2021-05-01:
Many thanks to all the developers and contributors for this great software!

Adding my own 2 cents to this feature request:
Much of what I design and print needs dimensional accuracy, not to the micron level, but as good as I can reasonably get without spending a month tweaking a single design.

I have been adding scaling factors (like an ABS shrinkage coefficient) to important XY dimensions of my models, but even when using global scaling factors with parametric design, it is tedious and I always forget to apply it to one (usually several) dimensional constraints in the CAD.  And then I have to adjust that global scaling factor and re-export all STLs depending on what type/color/manufacturer roll of filament I have in the printer, or want to print the same object in different materials/colors/etc.  CAD software has various ways to do scaling (each with it's own pros/cons), but I have to redo/re-export everything again depending on filament, so same issue.

I get that there are "Scale factors" in the main UI, but at least in the version I am using there are not nearly enough decimal places to adjust for dimensional accuracy.  I get that there is "Print Settings->Advanced->XY Size Compensation", but I'm not sure how to translate a raw unit-less scaling factor into "mm's", and scaling factors for ABS vs PETG are much more of a "filament setting" than a "print setting".

Perhaps adding to "printer settings":
- Checkbox: "Enable auto pre-scale of objects by selected filament"
Adding to "filament settings"
- Percentage number field, with plenty of decimal places: "Pre-scale factor for dimensional accuracy compensation"

Or something like that...

If the selected printer type is an MMU and a composite object uses multiple materials, then display a warning and turn off auto-scaling.  Or use an average scale.  Or just leave it to the user to adjust settings appropriately, which is probably the better option for MMU users anyway.

Just my 2 cents.  I hope filament shrinkage compensation will be added soon since doing it in CAD for every individual design and every desired filament gets painful after a while.  This feature has been added to other slicer software and apparently it does a decent job improving dimensional accuracy for individual filaments.  Thanks

alexandrkut, 2021-11-16:
I support the transfer of this option to filament tab.

neophyl, 2021-11-16:
Personally I do NOT support that change.  That setting is not only used to compensate for filament shrinkage.  It is also used quite a lot to make models with unprintably small parts printable.  Especially artistic models/sculptures.  Adding on 0.1 can make small details sliceable.

Also setting an absolute value is the wrong approach for shrinkage as generally it's a % value needed.  I'd prefer specific new settings added to the filament tab to combat the shrinkage problem.  Possibly something based on  the Super Slicer implementation or similar.

peterjhl, 2022-05-21:
I support this proposal. Maybe it should be put under filaments overrides?

jowgn, 2023-02-15:
Are there any plans to add this feature?

MrBlackmidi, 2024-02-08:
Hi, is there an update on about implementing this, also z shrinkage aswell?

Phil1988, 2024-03-17:
Just include this: https://github.com/prusa3d/PrusaSlicer/issues/11183 

And everyone would be happy.

6ffm70, 2025-01-08:
[image]

I found this in 2.8.0, but no documentation for it, except the popup help. 

[image]

What does it do, and how can I check in the interface that it will work?

closed 2026-09-29 by prusaslicer-bot[bot]