prusa3d/PrusaSlicer #3060: Support in Height Range Modifier not working

joeynuggetz, 2019-10-12:
### Version
2.1.0 
### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa MK2.5s

### Behavior
Adding a height range modifier and then selecting support from the drop down does nothing. I was hoping to be able to specify different support levels per layer but doesn't seem possible as nothing happens when selecting support materials. Is this broken or not yet implemented? 

_Is this a new feature request?_
Not really unless this is isn't what this feature does then it would be nice to set different support densities per layer.

mroek, 2019-10-20:
I just discovered this as well. Bummer. If it's not meant to work, then the menu option must be removed.

YuSanka, 2019-10-21:
Thanks for your report. This bug is known and already fixed with [commit](https://github.com/prusa3d/PrusaSlicer/commit/fdb51912c2c884c8ec0def6f5718c8b17e9df74c) .

closed 2019-10-21 by YuSanka

nordurljosahvida, 2019-10-24:
Thank you so much @YuSanka . Can I ask you how I can make use of this fix without waiting for stable release? I've been having issues building. https://github.com/prusa3d/PrusaSlicer/issues/3099

nordurljosahvida, 2019-10-24:
Ok, sorry, I'm confused, this commit does seem to be prior to 2.1.0 release, I therefore think it's still an issue on 2.1.0+linux64 stable.

nordurljosahvida, 2019-11-01:
@YuSanka i just managed to build HEAD locally, the option has been removed I see. Is it possible to reopen this issue as a feature request? @joeynuggetz are you still interested in the feature as I am?

3Dietrich, 2020-04-20:
? fixed, closed?
adding "Support material" to the Height Range does NOT work in 2.2.0+
I guess closed reports aren't read or re-fixed so I write a new one..

YuSanka, 2020-04-21:
Hi, guys. Sorry for the late answer. I missed conversation after the issue was closed. And it is my fault, that I didn't explain the fix. The thing is we don't support "Support materials" setting overriding for parts/modifiers and layer height ranges. It is implemented just for objects. And the bug was that the "Support materials" item was appeared in the context menu for parts/modifiers and layer height ranges. This bug was fixed.

3Dietrich, 2020-04-21:
well... if you mean that special hiding, okay, fixed. But you understand my problem: Support for Height Range is **still there** as an option. 
Anyway, good that I wrote [another bug](https://github.com/prusa3d/PrusaSlicer/issues/4100)

YuSanka, 2020-04-21:
> well... if you mean that special hiding, okay, fixed. But you understand my problem: Support for Height Range is **still there** as an option.

You are right :) It was in "Add Settings" list. But now it's fixed with [commit](https://github.com/prusa3d/PrusaSlicer/commit/9bc96bf28e42ba6959283fbb731e23bb878cf579)