prusa3d/PrusaSlicer #3061: Incorrect number of interface layers for >1

vintagepc, 2019-10-12:
### Version
2.1.0

### Operating system type + version
OpenSuSE 42.3 x64

### 3D printer brand / version + firmware version (if known)
Mk3S/MMU2

### Behavior
PS adds one too many interface layers if the value is greater than 1 for support material. 
(e.g. 2 adds 3, 3 makes 4). 1 is correct, which means there is no way to select exactly 2. This creates unnecessary/undesired toolchanges when using a different material for the interface layer (since 2 layers can be done with a single toolchange)

Picture: (Value is set to 2 in the support material settings:
[image]

3MF attached.

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs

[tens_carry_levers.3mf.gz](https://github.com/prusa3d/PrusaSlicer/files/3720772/tens_carry_levers.3mf.gz)

bubnikv, 2021-03-18:
Is this problem of yours still valid with 2.3.0?

I have tested with the current master + some refactoring yet to be committed and it seems fine.
[image]

bubnikv, 2021-03-18:
That image applies to two interface layers.

BTW I am doing some major refactoring of the supports, thus things may improve or break. I know you like to live on the edge, so you are compiling and running the master time to time.

vintagepc, 2021-03-18:
Yes, this definitely still happens for me. For example:

bubnikv, 2021-03-18:
> Yes, this definitely still happens for me with the official 2.3.0.

I mean current master. I have yet to commit one big change though.

vintagepc, 2021-03-18:
Ah, sorry, I was responding to this: 
>Is this problem of yours still valid with 2.3.0?

Happy to re-check against current master when you get everything in order.

bubnikv, 2021-03-22:
It is in the master.

vintagepc, 2021-03-22:
Yes, it does look like it has since been fixed! :+1:

jmsaltzman, 2021-04-28:
I'm getting hung up on this now: using 2 gives me 3 :(  This is 2.3.1, it was also there in 2.3.0.
[image]

vintagepc, 2021-04-28:
Beware that the topmost interface layer may not actually be one, it is often just an additional loop around the outside of the part but not actually under it. 
You can see this in my last screenshot, the number was set to 4, but there are 5 "thick" dark green interface layers and one thinner "decorative?" interface layer.

jmsaltzman, 2021-04-28:
Thanks! Those are actual interface layers though: I'm using PLA (gray here) as support for PETG, (orange), and it prints three real PLA layers.

vintagepc, 2021-04-28:
Fair enough - I wonder if this is a side-effect of it being soluble mode with 0 separation.

vintagepc, 2021-04-28:
Nevermind - I think the issue is that particular commit may not be in 2.3.1. 

I'm comparing master to the tag and it looks like there are a large number of support-related commits we probably won't see until e.g. 2.4.0

https://github.com/prusa3d/PrusaSlicer/compare/version_2.3.1...master - look around Mar 19th where lots of support refactoring happened

bubnikv, 2021-06-03:
@vintagepc thanks for heads up. closing.

closed 2021-06-03 by bubnikv

JonBerlin, 2021-12-09:
This is still an issue in 2.3.2!

lukasmatena, 2021-12-10:
@JonBerlin
As @vintagepc stated, the
> related commits we probably won't see until e.g. 2.4.0

It is broken in 2.3, fixed in 2.4. You can try in 2.4.0-beta3.