prusa3d/PrusaSlicer #3062: 3D-Labprint Models aren't sliceable (till years)

TomNbgDE, 2019-10-13:
More than two years I'm using Prusa-Slicer - till today it is not able to slice RC-Models from 3DLabprint which is sad - I have to use S3D for it, which I won't do anymore. As Cura, MH and S3D can do this - why can't Slicer it?

It would be nice to see in a next version!

mroek, 2019-10-13:
Are you able to provide a sample model that can be tested by others? I know the models from 3DLabPrint all cost money, and you can't expect the developers to purchase a model to test this.

Perhaps you can take a model and cut out a portion of it as a sample? It might still violate the license from the authors, but it would at least not be useable for anything except reproduce the issues.

That said, have you really exhausted all options in PrusaSlicer to get it to slice this? The most important setting would probably be the extrusion widths, but without a model to test with, you cannot expect to get any proper help on this. Sorry.

TomNbgDE, 2019-10-13:
Sorry - for sure:
You can find official test files from 3DLabprint here (I just link it)

[https://www.thingiverse.com/thing:2209989](url)

neophyl, 2019-10-13:
These models were discussed previously over on the Prusa Slicer forum.  There might even be an issue already here about them somewhere but the forum thread is located here https://forum.prusaprinters.org/forum/prusaslicer/slicing-rc-plane-wing/

One of the developers posted in there so its worth a read.

Jebtrix, 2019-10-14:
For a website calling itself 3Dlabprint you would think they could provide manifold models. Outer skin has no thickness, that makes it a non-manifold boundary feature. I've seen Cura mangle these models too trying to "repair" it to manifold. Is what it is I guess.

TomNbgDE, 2019-10-14:
You're right, that the kind how the models are built is not optimal. And I assume, it may have been a deal with S3D, that it could handle it without problems. But as Cura and Matterhackers is also usable - and the Problem with Slicer exists for more than 3 years now - my hope was, to implement the neccessary routines to slice it also.

I have so pretty optimized profiles for my mk2.5s and MK3s each with BMG and can't use them. The results are much better, than with S3D. The delivered basic-gcodes don't work well with BMG Extruders...

pstasek, 2019-12-27:
Hi, I'm writing here on behalf of 3dLabPrint. First of all, the models are made like this on a purpose, to be able to print lightweight and strong structures. The internal trusses are not connected to the shell, so it's not pronounced to the surface while it still connects and strengthens the airframe. The shell is printed in a single pass, then the nozzle continues to print the internals. We're using Simplify3D in a way, it wasn't probably originally meant to. Finally, we managed to amend the parts to be sliceable in Prusa Slicer with the internal structures as well by flipping the triangle normals for the outer surface. however, the prints are still far from optimal. In other slicers, we're able to control the gap between internals and outer shell, but in PS none of the parameters seems to affect this behaviour. The simulation looks promising though.

[image]
[image]

In case we found a solution to this issue, the Prusa Slicer might become fully supported by 3DLabPrint. I'll be happy to provide any info needed.

pstasek, 2019-12-27:
Just for the reference, we need to slice it like this: 
[image]
The internal truss blends into the outer shell in Prusa Slicer
[image]

We only use outer perimeters for this.

strongbow1, 2021-01-06:
Wow still not possible. I have models from them i bought 2-3 years ago and still its not sliceable in Prusa Slicer.

TomNbgDE, 2021-01-06:
I went over to IdeaMaker - there it works fine now...

strongbow1, 2021-01-06:
Thats not the point!! The point is they should get their act together and fix this issue. Im not going to a different slicer. They need to realise their customers and a lot of them are using prusa slicer

pstasek, 2021-01-06:
The point is, we do know, how to make an STL compatible with Prusa Slicer, but it introduces more minuses than pluses. (There's a P-47 Slic3r edition, that showed up as a dead-end in the workflow). We need the Prusa Slicer to update the features as mentioned above, then it would make sense to make the STL files compatible, as mentioned above (still a lot of manual work to do though).

strongbow1, 2021-01-06:
That brings me to my next question. Have you as a company sat down and talked to Prusa about the issue to bring a solution. You are selling to a big group that cant slice this. And it should of been top priority. When one looks at this post you start to question if you actually care for customer satisfaction or money?! And looking how old this is one can assume no communication is flowing between 3dlabprint and Prusa. Which brings me back to customer/money situation!

pstasek, 2021-01-06:
Although it would be nice if our models would be finally sliceable in Prusa Slicer, we and our valued customers already have multiple tools to slice our models. Why it should be a top priority then? We have no leverage on Prusa Slicer developers to address this issue. Prusa Slicer is a great piece of software for other tasks, it just doesn't suit our specific needs until the above is solved (might even be a very complex problem or Prusa Slicer users/developers may have different priorities). Thin-wall printing is a very different discipline than what is printing of standard manifold models with very different goals, so you still need to set up your profiles from scratch anyways no matter what slicer you want to use. There's no perfect slicer for everything. You don't use a hammer as a screwdriver either. And furthermore, the tools required to slice our models are mentioned in the description before you hit the trigger.

strongbow1, 2021-01-06:
So its confirmed there is no communication between the 2 companies. That would of at least been the minimum 3dlabprint could have done. Because if prusa does not know there is a problem how can you fox a problem. You can take a hammer but if you don't know you need a screw driver then well the problem is communication. That is what is happening here. Open a channel between 3dlabprint and prusa and you will be surprised that there might be a simple solution to a difficult problem! So yes it should at least be a priority.

pstasek, 2021-01-07:
I'm sorry, but this is an open Prusa Slicer Github issue (feature request), not anonymous discussion somewhere in the forum. This is the way, how you communicate with Prusa Slicer (or any other) developers in the world of open-source software. I described the problem above and when any of the developer address this, I'm here to provide any additional info required to solve this. As you can see, we're giving it some priority, as my reaction came within a couple of hours from your post. Let's not spam the thread further without advancing to the solution.

rupert80, 2021-01-10:
@strongbow1 - as another example of models being created for a particular slicer, the Planeprint models are even more restricted than the 3DLabPrint ones. Planeprint models can only be sliced with Cura as it's the only one which can print single perimeter internal structures. Are you going to yell at them for not supporting Prusaslicr or S3D?

strongbow1, 2021-01-16:
Will this issue be addressed? Since several versions have been released from 2.1.0?

bubnikv, 2021-01-18:
@strongbow1 
> Will this issue be addressed? Since several versions have been released from 2.1.0?

The 3DLabPrint models are invalid models from the point of computer solid geometry modelling. We would have to implement a special mode for handling such broken models.

There are 1800 open issues. Which one has a higher priority?

pstasek, 2021-01-18:
@bubnikv 
Thanks for the follow-up. I totally understand the priorities, however, if you need any cooperation from our (3DLabPrint) side to handle this type of models, please let me know, I'll be happy to provide any files needed to develop this feature. 

Just please don't call the models "broken" as it's designed like this for a purpose ;-)

pstasek, 2021-01-18:
@bubnikv 
If it comes to the priority, please also check my above posts, with reversed shell surface normals the paths are sliced in the order we need, the only issue seems to be the tolerance between the outer shell and internal ribs. Reversing normals is something we can do if it helps the cause. The designed gap size between the outer surface and the internal ribs is required and comes from the nozzle diameter (extrusion width).

pstasek, 2021-01-18:
Adding the test file for the B25J wing - standard STL, STL with reversed normals and a PS 3mf file with the best PS result I was able to get so far.
[B25J_wing_PS_test_file.zip](https://github.com/prusa3d/PrusaSlicer/files/5829200/B25J_wing_PS_test_file.zip)

strongbow1, 2021-01-18:
@bubnikv Well it needs to have a higher priority. I am sure this is not the youngest issue open. But yet it has been ignored for so long. That is the surprising part that Prusa has not investigated this earlier and are seemingly ignoring open problems..

pstasek, 2021-01-18:
@strongbow1 I'm pretty sure, this is not the best way, how to ask a developer for a favour to implement a new feature... this is not a bug, it's a feature request, so let's be polite. Neither Prusa Slicer nor we are claiming the Prusa Slicer can slice our models at the moment, although it would be much appreciated by the community.

bubnikv, 2021-05-28:
@pstasek please send me a personal e-mail to vojtech.bubnik at prusa3d cz
I will get in touch with you.

For PrusaSlicer 2.4 we have reworked slicing of objects with multiple parts, it should respect the "clip multi-part objects" settings, so if you manage to load the wing shell and the internal structures as separate volumes of an object and you unset the "clip multi-part objects" settings, PrusaSlicer should slice these parts independently even if they overlap. That is the theory. Let's evaluate it for your models.

strongbow1, 2021-05-28:
Its a very complicated way. Prefer to load it in and just hit slice! None of these clip separate volume stuff. Why make it so complicated? Easy 1 Load the part in 2 slice the part 3 print. Don't want to spend time on looking for settings fine tuning these objects and volumes! Keep it simple thats how it should be

bubnikv, 2021-05-28:
@strongbow1 
By default PrusaSlicer merges overlapping volumes. That is simple and correct as it should be. Unfortunately that does not work for models that were designed to overlap, but to be sliced independently.

If you want to make it simple as it should be, you need to load a PrusaSlicer project that already has all the settings adjusted for the particular model.

strongbow1, 2021-05-28:
By default other slicer can manage this. By default Prusaslicer can not! So what is the difference between other slicers? They actually update their slicers to work! This issue has been open for 2 year or more and nobody in this time did anything to remedy this. So yeah its a bit disappointing

bubnikv, 2021-05-28:
> By default other slicer can manage this.

I know that for example S3D cannot handle intersecting and self-intersecting bodies correctly. S3D shall unite these intersecting bodies into a single body, but it does not. S3D extrudes one body over the other creating a mess, clogging your nozzle, wasting your time and material. What a shame! And they are not doing anything about that! They are not listening to their customers! They are completely ignoring them! I would never buy a product from them!

strongbow1, 2021-05-28:
S3D can handle 3D-Labs model just fine. Way better than Prusa slicer can at this moment. And I never had any issue with S3D and all my other printers. I am using prusa slicer only on my prusa printers. And that in 2 years they could not Adress this issue is a real indication that they also dont listen to their customers!

pstasek, 2021-05-29:
> @pstasek please send me a personal e-mail to vojtech.bubnik at prusa3d cz
> I will get in touch with you.

E-mail sent, thanks for addressing this.

bubnikv, 2021-06-01:
I had a call with @pstasek yesterday. After the call I investigated the models provided and I think I understand the issue now. S3D by default uses an Even-Odd filling rule, while PrusaSlicer and Cura uses the Positive filling rule by default. Cura has a switch available to switch to the Even-Odd filling rule.

[image]
http://www.angusj.com/delphi/clipper/documentation/Docs/Units/ClipperLib/Types/PolyFillType.htm

strongbow1, 2021-06-01:
So is there a fix coming out soon for this problem FINALLY? Or are we going to wait another 2 years for something?

liborvanek, 2021-06-01:
> So is there a fix coming out soon for this problem FINALLY? Or are we going to wait another 2 years for something?

Will you stop trolling? Let the devs work. I'm quite happy they're addressing this issue and being mean doesn't help at all.

strongbow1, 2021-06-01:
Its been more than 2 years! Its my right to ask if there will be a fix! If you dint like it scroll along!

neophyl, 2021-06-01:
Its free software that you can use as you want.  Did you pay for it and its not doing what was advertised ?  No then you don't have a right to demand anything.  And that's all you have been doing, demanding.

bubnikv, 2021-06-01:
I have added a new "Slicing Mode" parameter to switch between the Regular, Even / Odd (aka S3D or 3DLabPrint mode) and "Positive". The "Even / Odd" rule is the default and only rule applied by S3D. Cura applies a "Positive" rule by default, it has to be switched to "Even / Odd" to slice the 3DLabPrint models correctly.

[image]

[image]

strongbow1, 2021-06-01:
Its a software that got delivered with my mk2 and mk3. So my purchase of these 2 printers directly support the development of Prusa Slicer. You also you need to brush up on what rights one has!

strongbow1, 2021-06-01:
> I have added a new "Slicing Mode" parameter to switch between the Regular, Even / Odd (aka S3D or 3DLabPrint mode) and "Positive". The "Even / Odd" rule is the default and only rule applied by S3D. [...]

When will this be available for us to download?

bubnikv, 2021-06-01:
Implemented with 20ba7c0a1f934430ecdd300bb25f2377e552cc38

New parameter "Slicing Mode" for supporting 3DLabPrint airplane models.
S3D's strategy for merging self intersecting models is "Even / Odd"
which PrusaSlicer now supports as an alternative to "Positive" rule.
Also added a "Close Holes" option to fill in all internal structures.

The new parameter will be released with PrusaSlicer 2.4.0
Closing.

closed 2021-06-01 by bubnikv

liborvanek, 2021-06-01:
@bubnikv Thanks a lot, much appreciated! 👍

bubnikv, 2021-06-01:
@strongbow1 Sehr geehrter Herr Sandro Strimer,

On the left side you see the result of applying the Even / Odd rule (S3D default) onto the famous Tree Frog model. The Tree Frog model is known to contain self intersections. The Even / Odd rule applied by S3D leads to holes in the model inside the self intersections. On the right side you see the result of the PrusaSlicer / Cura default strategy, which fills these holes.
[image]

You see, one shoe does not fit all. We believe that for the majority of models the default PrusaSlicer strategy is better than the S3D strategy. Therefore both PrusaSlicer and Cura need the user to switch a slicing strategy for the 3DLabPrint models.

strongbow1, 2021-06-01:
They look the same to me. Which holes? Its hopefully soon 2.4 will be released and we finally will have the option. Which we previously did not have.

pstasek, 2021-06-01:
Many thanks, @bubnikv for adding the Even/Odd slicing mode. I'm about to tune up a 3DLabPrint specific MK3 3mf file, where the STL could be easily dropped on the bed and as soon as the new version comes out, the Prusa Slicer is becoming an officially supported platform for our models.

bubnikv, 2021-06-03:
I have verified that Cura uses the same slicing mode by default as PrusaSlicer. IdeaMaker uses the Even / Odd rule by default. Indeed, from the UI perspective it is a copy of S3D.

rtyr, 2021-06-03:
@strongbow1 

> They look the same to me. Which holes?

[image]

strongbow1, 2021-06-03:
> @strongbow1 
> 
> 
> 
> > They look the same to me. Which holes?
> 
> 
> 
> [image]
> 
> 
> 
> 

Those don't look like holes. I was physically looking for holes and not at the outlines that are a bit different. But thanks for pointing it out.

rollyceballos, 2021-07-02:
This isn't enabled in 2.3.2 beta?

bubnikv, 2021-07-02:
no. It will be in 2.4.0. We are getting close to an alpha.

Le0Michine, 2021-09-20:
I just tried to slice 3DLabPrint wing in PS2.4 and the results are looking promising. Besides turning on Even/Odd setting for slicing mode I had to disable "detect thin walls" otherwise I was getting artifacts on the surface (refer the pictures below). After all that I get same picture on preview as from cura which is a very good sign. @bubnikv thank you for looking into the issue!

Detect thin walls enabled
[image]

[image]

Detect thin walls disabled
[image]

eskimgv, 2024-08-19:
> When will this be available for us to download?

Thank you very much, that's the solution that i looking for years :) but just in case i had to decrease the "slice gap closing radius" also.