prusa3d/PrusaSlicer #3063: Visualize unsupported Islands

mylife4aiurr, 2019-10-13:
PrusaSlicer SL1, feature request

Need a way in prusaslicer to identify unsupported islands in prusaslicer. This way users can see where to place supports. Auto supports is an option but something it may miss certain details. Or maybe a user would like to place their own support or add additional supports when using auto supports.

I would like to prusaslicer to shade differently portions of model that need to supported while in the support placement function

nigeljohnson73, 2020-04-10:
You can autogenerate. then manual edit and select the box "Lock supports under new islands", then ctrl-A to select all supports, then press delete to delete all non-locked supports, and there you have all islands identified and supported... so you can add your own additional supports.

mylife4aiurr, 2020-04-10:
Thanks for your response. I'm aware you can use auto support.
I don't rely on prusaslicer to pick best orientation for my model any more.
I don't rely or trust prusaslicer to auto support my model better then I've learned in these few months. 
I place all my supports manually in another free software because it does show unsupported islands and the actual support not just a dot.
Editing dots is just a stone age way to do it.....

You don't see how the supports interact with the bed or other parts of the model, because prusaslicer only shows you dots at the point support contacts model.

Humor me and watch these videos:
https://www.youtube.com/playlist?list=PLC4QzbTxrEF-fA1OC_-zE_4f6n1K-w2SS

I guarantee it will change your mind, prusaslicer hollows better but custom support placement is a frustrating mess.

mylife4aiurr, 2020-04-13:
https://youtu.be/3ykwlg5kF-8

closed 2026-09-29 by prusaslicer-bot[bot]