prusa3d/PrusaSlicer #3064: Make PrusaSlicer downloadable from Pop Shop on Pop OS

I-like-stuff, 2019-10-13:
I think it would be nice to be able to download PrusaSlicer from the bundled package manager in Pop OS as it would make it easier to update and then you wouldn’t have to mess around with each the AppImage files.

Jebtrix, 2019-10-14:
I couldn't find anything anywhere about submitting an app for their gui package manager. You'd have better luck asking them. Their package maintainers are responsible for this, not the original developers. Only exceptions I know of are snap packages for Ubuntu and Elementary OS's software store.  https://github.com/pop-os/pop

I-like-stuff, 2019-10-14:
Pop Shop is based off of Elementary OS’s package manager.

I-like-stuff, 2019-10-14:
This is what someone told me:

Firstly the program needs to be in the Pop!_OS repo, then you will need to create a DEP-11 YAML for the AppStream metadata.

xarbit, 2020-04-09:
I started to create prusa-slicer as a FlatPak and will submit it to FlatHub once completed (actually it is currently building). 

Pop Shop/Pop OS support FlatPak & FlatHub .. This issue/ticket is basically a duplicate of:   

https://github.com/prusa3d/PrusaSlicer/issues/1124

xarbit, 2020-04-10:
The application is submitted to FlatHub and is awaiting review. 
If this is something the Prusa development/release wants to have control over please let me know. 

Feel free to try the flatpak bundle from my repository:
https://github.com/xarbit/com.prusa.PrusaSlicer/releases/tag/v2.2.0-flatpak-testing-1

From my view, now I can finally use PrusaSlicer with my new Fedora Silverblue workstation connected to my MK3s printer :-)

I-like-stuff, 2020-04-12:
I have since moved on from Pop os and moved to Manjaro, an Arch Linux based distro which has access to the AUR which has PrusaSlicer that is currently well maintained it only took a couple weeks for PrusaSlicer to be updated to the current release. However I think having it on flakhub would be beneficial for many more people as flakhub is becoming much more popular and used in more distros. Having applications in one easy place where they get updated with other packages is much better than having to re-download the AppImage. This is one reason I like manjaro/arch as there are so many packages for applications that other distros don’t have. (Sorry for sorta turning into a Linux review but it gives more reason for things similar like this to become more common)

xarbit, 2020-04-27:
flathub has accepted the PrusaSlicer application, it should arrive in the flathub channel for anyone to use shortly.
 
For the time being I will be the maintainer of this package until upstream (Prusa) is willing to take over this (which should be the goal). 
  
The manifest for this can be found:
https://github.com/flathub/com.prusa3d.PrusaSlicer

Dear Prusa Team, if you need write access and want to take ownership, just open a ticket and FlatHub will take care of this.

I assume this also means this ticket can be closed.

I-like-stuff, 2020-04-29:
That's great, I have now moved to Arch which has Prusa Slicer in it's repos so don't need to use flathub. However it would be great to see if Prusa take this over and make it better known to people as messing with app images on Linux can be a bit annoying when having to update it separately to other packages.

closed 2020-04-29 by I-like-stuff