prusa3d/PrusaSlicer #3067: Arbitrary filament overrides

haarp, 2019-10-14:
PS 2.1.0

Hello,

I would like to feature request that filaments can do arbitrary overrides, not just retraction rleated ones. There's always some exotic setting that needs to be changed for certain filaments, e.g. TPU needs more Z distance between supports and and the object.

The .ini file format should be easy, as each override already starts with the `filament_` prefix. It's just a matter of handling in the UI. How about a button next to each setting labelled "make filament override"?

Thanks!

sticilface, 2020-04-02:
I've left a comment on New feature request - Additional "Filament Settings" for retraction #480, but there are numerous requests for filament overrides.  I too need custom speed settings for different filaments.  namely TPU. Would love to see a more flexible approach.

bubmann, 2021-03-13:
I second the request, my specific problem would be infill/perimeter overlap. my PETG on 0.8mm only needs 20%, my PLA more like 30% or more. I guess I could use 25% for both ... but I'd rather be pedantic about that

rozhkovets, 2023-11-18:
I also want this feature. It so happens that I often forget to change the print speed when I print TPU and PLA

closed 2026-09-29 by prusaslicer-bot[bot]