prusa3d/PrusaSlicer #3068: Printing two overlapping objects separately

mroek, 2019-10-14:
### Version
2.1.0

_Is this a new feature request?_
Yes, I think so

Sometimes I want to print an object with more than one color with my single (non-Prusa) extruder printer, but the two objects overlap in height. More specifically, the base objects has a hollow section that I want to print into with a different color. Or I want to print the contents of the hollow section first, and then print the base on top of this.

I am used to Simplify3D, where I can do this easily by simply assigning separate processes to the two objects, and then print them separately, and not moving the build plate between the prints. Proper Z-hop must also be used to avoid that the nozzle crashes into what's already on the build plate.

It feels like I should be able to do this also in PrusaSlicer, but the key thing I am unable to achieve is to align the two objects properly without also being forced to print them in one go. So with that said, here's an example:

[image]

This is how it looks with the two objects loaded separately into PS, and then just auto-arranged. I can't find any method of properly aligning these together, manually adjusting is not acceptable, since these are exported in their proper locations from CAD.

So then I tried the method of loading the base part first, then adding the inlay as a part, and used place on face to lay it flat. This produces exactly what I want, where the inlay is positioned correctly inside the cutout in the base:

[image]

However, now there is no way for me to print these separately while keeping their relative positions. I want to print just the base part first, and then the inlay, which then looks like it starts to print in the air (but in reality it is on top of the already printed base, of course).

The requested feature is therefore:

_a) A command to align (and perhaps group) selected objects according to their relative positions in the STL files and to print them in exactly those posisions (not auto-dropping to bed or something)._

----or----

_b) Allowing individual control of print visibility for parts and subparts, so that you can choose which parts to slice and print, and again, no automatic dropping to bed for parts that appears to be printed in mid-air._

I've also attached the sample model (as shown in the images) to this request.

[model.zip](https://github.com/prusa3d/PrusaSlicer/files/3723427/model.zip)

mroek, 2019-10-14:
> 
> 
> I think there is option to do this if you slice this model in multi extruder profile with M600 command in custom toolchange gcode. When you replace Tx command for extruder switching by M600 for manual filament change. I hope I wrote it understandably..

I don't think this allows for the objects overlapping, like in my example. The inlay is to be printed inside a shallow cavity in the base (inlay is just 0.6 mm thick in the example), but MMU mode is made for sequential printing of layers with different colors. That means the first toolchange will appear before the base is printed to it's final height, so I'd have to change filament more than once to get it done. That is tedious, and not really necessary, since the depth of the cavity is such that the nozzle can reach down just fine. I've done this multiple times with great success (but not with PrusaSlicer, obviously).

The same issue of course appears if I rotate the part so that the inlay is printed first, then the base on top (which is also something I have done in the past, but again with a different slicer).

FidelCapo, 2019-10-14:
I'm not sure it's a good solution. If the nozzle has a conical shape and if the inlay is higher than the layer height, the nozzle will interfere with the already printed object, right?

mroek, 2019-10-14:
The inlay needs to be kept thin, but it can certainly be higher than the layer height without any issues. In the example, I'm using 0.6 mm, which I would typically print as three layers (0.2 mm layer height). That works really well, no problems.

In many cases I would print the inlay first (against the bed), but the issue remains the same, since I then require the base to first print around the inlay, and then on top of it. Again, fully doable in the real world (and with excellent results), just not with PrusaSlicer.

Here's a Youtube video by Devin Montes where this very method is being used with great success: https://www.youtube.com/watch?v=KV2AjyowXX4

So as you can see, this is a viable workflow, and I think it should be reasonably easy to make it work also in PrusaSlicer, as in my inital description.

neophyl, 2019-10-14:
Isn't the end result the same as doing what Joan described in this thread -https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-how-do-i-print-this-printing-help/acting-as-my-own-mmu/  ?

That discussion / technique involves printing different colour text/logos inside a different colour using a manual mmu change.  If the inset layers are thin that practically there isnt much difference ?

mroek, 2019-10-14:
There is one crucial difference: What I am asking for requires just one single filament change, with the "manual MMU method", I would have to make a change for **_each layer_** where the base and the inlay overlaps. Now, that is of course possible to do, but it is more cumbersome and not necessary as long as the inlay is thin enough for the nozzle to reach down.

I have a feeling that this shouldn't really be too difficult to implement in PS, as all the needed code is already there, specifically the aligning of STLs that has a common origin. I just need a way to tell PS that I want to print each STL separately, and exactly where it is positioned (i.e disabling auto-drop to bed).

mroek, 2019-10-14:
One more thing: With my method I can print the inlay as 4 layers with 0.15 mm layer height (for better coverage), but with the manual MMU method the layer height can't be different between objects, so that means even more filament changes if I decrease the layer height for better coverage.

FidelCapo, 2019-10-14:
Ok, I tried one more workaround in PrusaSlicer. Here is repro:
Add base
Add inlay as part
Place on face on inlay (so inlay is on bed)
Add settings to base part and set 0 perimeters, 0 top and bottom layers and 0% infill
Increase Z hop
Slice and export gcode
Delete settings and add same settings to inlay
Slice and export second gcode

It will work because in this case there isn't any empty layers, but it work only if inlay is on bed.

mroek, 2019-10-14:
Thank you! That is creative (why didn't I think of that?), and it will work as long as the inlay is printed towards the bed, like you say. For a moment I thought it would also work the other way, because when I sliced it, it did show the inlay as being printed mid-air (in the preview), but unfortunately PS refused to actually export that g-code with an error message about empty layers.

I am still going to leave this feature request open, because I think there is a case to be made for this kind of usage, but the suggested workaround will work for the one specific case.

FidelCapo, 2019-10-15:
Ok. I think this is very similar request https://github.com/prusa3d/PrusaSlicer/issues/1513

mroek, 2019-10-15:
@FidelCapo Yes, that seems to be pretty similar, but it is not identical. While I agree with that request as well, my request is even simpler. I just want to be able to group parts to get their native alignment, and then print them separately with whatever settings I like. 

Judging from the developer responses in that other issue, it seems to be hard getting the message through to them, possibly because their workload on other issues and features are too big.

clinta, 2021-04-01:
I'm trying to solve this same problem. I think an elegant solution would be a `maximum height before toolchange` option. Then you could do this as normally, using different extruders for different parts. But the slicer would lay down multiple layers of one color, up to `maximum height before toolchange` before adding the toolchange gcode. Then after changing tools it automatically adds the necessary increase in z-hop height for the layers with the new tool.

wisco-kid, 2023-10-06:
does someone have a basic brake down to the g-code as I import it into Note++ Its looks crazyier than what I've seen before. and it seemed simpler when i used simplify3d.... and now I'am attempting to edit gcode from prusa slicer 2.6.1.
 Just curious

excalq, 2025-02-02:
Also wishing for a `layers before toolchange` feature, as expressed in https://github.com/prusa3d/PrusaSlicer/issues/8415. In my case, for printing flat items with a colored inlay, and wishing to print 2-3 layers at a time before having to change filaments.

closed 2026-09-29 by prusaslicer-bot[bot]