prusa3d/PrusaSlicer #3070: Feature Request: Multi section purge block like in e3D tool changer for better multi material printing

Gosga365, 2019-10-14:
### Version 2.1.0

### Windows 10

### MK3S MMU2S 3.8.0 (1.0.6)

### Feature request: modified purge block for multi material printing

Add an option to divide the purge block into discrete sections for each material used in the print. This would be a similar manner to the e3D tool changer slicing where one purge block with a small square for each filament in the block. This allows different materials to stick more easily to each other, like when printing with PLA and PVA or PLA and PETG which tend to have poor adhesion when blended in the default purge block setup.

bubnikv, 2021-06-03:
Does it mean that you would have to extrude all sections even though an extruder would otherwise not extrude in the object at that particular layer?

ssill2, 2021-06-03:
I'm thinking about getting a e3d tool changer because I was hoping to eliminate having to use a purge tower.  I thought with separate physical extruders the purge tower was effectively not needed.  I do agree that with the MMU this feature would be nice as I've personally seen the bad adhesion with different materials, like PETG and BVOH

Gosga365, 2021-06-04:
You would most likely need to print a sparse layer at least for every layer. So as an example, if you were printing an object where the bottom half is TPU and the top is PLA, you would have 2 pillars in the purge tower, 1 for each material. The PLA tower would be sparse layers for the bottom half just keeping it structurally sound and at the right height. Then it would go to being fully purged into in the top half while the TPU tower would no longer need to be printed or would have only sparse layers up until TPU was actively used again.

With a toolchanging printer, the need for discreet towers is much less, they are usually used to prime the nozzle and get good consistent plastic flow before resuming on the print and to catch any weird strings or drips before they become a part of your print too.

ssill2, 2021-06-04:
Ah that makes sense thank you.  I’m nearly ready to buy one but still doing research.  I’m totally done with the mmu

acekrystal, 2023-12-15:
Long time ago I wanted to use multi material for support using incompatible materials. Now that I have the XL I run in this whipe-tower problem too.

Every print its just wishing that the tower stay's together till the end of a multi material print of materials like PLA with PETG support.

Having an option to generate a whipe tower for each material, or having an option to section the whipe tower in material types would make the tower stronger and should help increase the usefulness of using multi materials for support.

ssill2, 2023-12-15:
which materials specifically?    I'm doing prints with PLA, PETG, Aquasys120(soluble) and TPU without too much issue.

ssill2, 2023-12-15:
I have a pretty ambitious 4.5 day print going with those materials above and the wipetower is holding together.

ssill2, 2023-12-15:
[image]

This is prusament jetblack petg, giantarm silver petg, and aquasys120.   other than a little stringing on the aquasys(printing at 238C), it's pretty good.   no issues with the wipe tower at all.   I've also done tpu and pla on the same wipe tower.  the XL does this a LOT better than the MMU ever did.  and the wipe tower is mostly hollow.

ssill2, 2023-12-15:
The one I was keen to test was using TPU to print the model, soluble(Aquasys120) as the interface, and the PLA as the support material since it's stronger than building talk supports out of tpu.   This worked fine.  I've not tried PLA + PETG though.  I do remember fighting with BVOH and PETG on the same wipe tower with the MMU.

acekrystal, 2023-12-20:
I do need to add, that I did not know yet you could set a primary nozzle for the structural (outside) part of the wipe tower... This makes me at least able to print my objects now (I see you are using the same setting. Before I literally had the full tower printed in different materials).

But even with this setting (Wipe tower extruder:) I find it still not really reliable. as you can also seen on my added images :
[image]
[image]

I'm printing a combo of PLA + NGen. And the Ngen does (intentionally) not wanna stick to PLA. Even with the outside of the wipe tower printed only in PLA, The infill with NGEN seems to not be stable enough, causing warp-up, nozzle collisions, droops, and out of place parts. Making for a very nerves technician (me). NGen behaves a bit like PET, but prints with better quality more like PLA. A full NGen printed part would normally not cause a mess like that, so I expect a better multi-material wipe tower should still increase reliability and more sleeping hours for the technician :).

ssill2, 2023-12-23:
now that I've spent several days the past week doing various materials plus solubles(aquasys 120 and bvoh) I've can definitely see a problem with solubles in general.    I got my test down to a 5 min print that has a layer of petg, then immediatley lays the soluble material on top of it for a few layers and then lays another layer of petg on it.   I also did some sainsmart tpu + bovh yesterday.  Waiting 8+ min for the "absorbing heat"  each iteration for a 5 min print is.... annoying.

Both bovh and aquasys 120 suffer from the same problem on the XL.  When the extruder is fetched each time and then it starts extruding on the priming tower, the first several mm don't appear to to stick and then magically it starts sticking, but that first several mm because problematic for things built on top of it.  I'm convinced this is because the material is still coming up to temp from the docked temp to the printed temp.  I played round with the aquays 120 by keeping the docked temp closer to the print temp and it did seem to help.

Coming from an MMU, it seems like the MMU sort of solved this problem by starting extrusion slow on the prime/wipe tower on filament load.  I need to dig some more into the available settings in the filament profile, but I found that increasing the amount purged onto the priming tower when the filament is first fetched tends to give it more time to come up to temp(and thus stick) but the initial extrusion suffers the issue of not sticking.

A couple of things I think could help and I intend to test.
1. Increase the docked temp to be closer to the temp it needs to print at.  This should allow it to be at temp immediately when it starts extruding to prime tower.  
2. find some way to slow down the print speed on the priming tower for this filament.

closed 2026-09-29 by prusaslicer-bot[bot]