prusa3d/PrusaSlicer #3075: [Regression] Reload from Disk disabled for any imported AMF files

Jebtrix, 2019-10-18:
### Version
Latest git build of master

### Operating system type + version
Windows 10 PRO 64bit

### Behavior
Imported AMF files cannot Reload from Disk but STL files still work fine.

@enricoturri1966 Broke at commit: 3a40565d03608c2440e772d58121ef16fc84744e

enricoturri1966, 2019-10-18:
@Jebtrix
Could you please add a repro steps list so that I can test and figure out in which way I broke it with the cited commit ?

Jebtrix, 2019-10-18:
Sure no problem.
## Steps

* Start PrusaSlicer
* Import .AMF part file
* Right-click on object and **Reload from Disk** is disabled/grayed out

## Test File
[Solidworks2018_Cube_Sample.AMF.zip](https://github.com/prusa3d/PrusaSlicer/files/3744715/Solidworks2018_Cube_Sample.AMF.zip)

enricoturri1966, 2019-10-21:
@Jebtrix
Could you please check if commit:  https://github.com/prusa3d/PrusaSlicer/commit/28a6bd7c3013ebf9c57353c600fc9a60f0f75ad2 fixed this issue ?

Jebtrix, 2019-10-21:
@enricoturri1966 Yup they did the trick. I use mainly AMF files so many thanks.

Jebtrix, 2022-01-21:
Latest git build of master @ Revision: 959804c80ce449b7ea008494f32a75a287204742

Its doing the same thing with AMF and 3MF object files. Create new project, import AMF or 3MF object, save then re-open. Reload from Disk invokes "Unable to reload" error message.

Jebtrix, 2022-01-21:
Moving

closed 2022-01-21 by Jebtrix