prusa3d/PrusaSlicer #3076: [Bug] GUI_Preview.cpp > Preview::on_combochecklist_features event is double firing

Jebtrix, 2019-10-18:
Win10 PRO 64bit
Latest git master

`void Preview::on_combochecklist_features(wxCommandEvent& evt)` @ slic3r/GUI/GUI_Preview.cpp

Poking around some code I noticed this was triggering two consecutive wxEVT_CHECKLISTBOX events for a single item click. Firing an unnecessary `refresh_print()` is minor but this did manage to throw a wrench into what I was trying to do so I figured I'd mention it.

bubnikv, 2020-03-20:
@enricoturri1966 would you please look into that?

enricoturri1966, 2020-03-20:
This is due to void wxCheckListBoxComboPopup::OnCheckListBox(wxCommandEvent& evt)
where the event gets duplicated and so processed twice.
If I remove the content of
    if (m_check_box_events_status == OnCheckListBoxFunction::FreeToProceed )
statement, it works fine on Windows (a single call of Preview::on_combochecklist_features() happens)
but I don't know what would happen to other platform due to the other code around it, which I did not write.

It may be that the
    evt.Skip();
in the method is enough to have it propagated to Preview::on_combochecklist_features() without the need to duplication.

In any case, the double call to Preview::on_combochecklist_features() has not a big impact, no toolpath regeneration happens.

rtyr, 2021-04-28:
Should be fixed. Closing.

closed 2021-04-28 by rtyr