prusa3d/PrusaSlicer #3077: Unexpected Result from Slicing

AndreasRothmann, 2019-10-18:
### Version
PrusaSlicer-2.1.0+win64-201909160915

### Operating system type + version
Windows Version: Windows 10 (build 18362), 64-bit edition

### 3D printer brand / version + firmware version (if known)
Prusa MK3, most current firmware - but it's a Slic3r not a printer problem.

### Behavior
I exported a quite complex model from TurboCAD 2018 as a 3mf and as a stl file. Both files create the same problem. When importing the files, the model looks good. But when I start slicing, I get a preview in which Slic3er is confused about the orientation of parts of the outline.

#### Project File (.3MF) where problem occurs

[ProblemWithSlic3r.zip](https://github.com/prusa3d/PrusaSlicer/files/3743786/ProblemWithSlic3r.zip)

[image]
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

Jebtrix, 2019-10-18:
Its a bad non-manifold mesh. You can try Netfabb to repair it. Even just in the Editor view of PrusaSlicer you can tell there is geometry issues.

[image]

AndreasRothmann, 2019-10-21:
I fixed the model with 3D Builder - and after that it worked. But there was also a logical  issue in my model: The geometry was not as I intended it to be. After I fixed that, Slic3er had no problems with the new STL file. Tnanx for your support!

I used TurboCAD to design this part. The software seems to have problems in generating valid STL geometries. I'll open a ticket for TurboCAD.

closed 2019-10-21 by AndreasRothmann