prusa3d/PrusaSlicer #3078: automatic CTE compensation

lukeskymuh, 2019-10-18:
### Version
2.0.0

### Operating system type + version
Win 10

### 3D printer brand / version + firmware version (if known)
i2 MK3

### Behavior
Hi,
it is known that prints shrink due to the temperature change from printing to room temperature due to their coefficient of thermal Expansion (CTE). I have analyzed this and this is main dimensional erreor for my MK3.
It would be great if the slicer would have a function to scale your prints in dependancy of the CTE of the material and the heatbed temperature.
Do you think this is possible to implement this feature?
Some links about this topic:
https://www.thingiverse.com/thing:2484766
https://www.youtube.com/

_Is this a new feature request?_yes

MStylesMS, 2020-10-30:
I agree with this request.  Having a default shrinkage factor for each filament profile based upon the middle range of the range of the printing temperature vs a typical ambient temperature (say 22.5C).  This would make sense one the Advanced page of the filament definition.

neophyl, 2020-10-30:
This has been implemented in Super Slicer.  I don't know how well it works as I have not used it as yet but its there as one of the filament settings. 
https://github.com/supermerill/SuperSlicer

manolitto, 2021-01-13:
> This has been implemented in Super Slicer. I don't know how well it works as I have not used it as yet but its there as one of the filament settings.
> https://github.com/supermerill/SuperSlicer

SuperSlicers "Shrinkage" feature only auto-scales in x and y direction but not in z !? At least that is how it is documented. This makes no sense IMHO if you want to compensate CTE.

Simplest solution for CTE compensation:
a new "scaling factor" filament setting which auto-scales printed objects in x, y and z.

Most flexible and future proof solution for CTE compensation:
three "scaling factor" filament settings - one for each axis

manolitto, 2022-03-24:
> > This has been implemented in Super Slicer. I don't know how well it works as I have not used it as yet but its there as one of the filament settings.
> > https://github.com/supermerill/SuperSlicer
> 
> SuperSlicers "Shrinkage" feature only auto-scales in x and y direction but not in z !? [see here](https://github.com/supermerill/SuperSlicer/issues/287) At least that is how it is documented. This makes no sense IMHO if you want to compensate CTE.
> 

I learned that SuperSlicer is doing right by just compensating in x and y. Reason is: the z size is given by the vertical steps which are always exact. So, every single layer is printed exactly at the height it should be - regardless of the shrinking of already printed layers.

So I propose this instead:

Simplest solution for CTE compensation: a new "scaling factor" filament setting which auto-scales printed objects in x, y

Most flexible and future proof solution for CTE compensation: one "scaling factor" filament setting for x and y compensation

Gefionious, 2022-03-31:
This I think is the last remaining feature I love with Simplify3D although I have transitioned 99% of my work to PruseSlicer. They have both an import model script feature as well as a dimensional XY correction in the setup. Is a great feature to add for the overall printer (if I consistently print +-) but now I am printing ASA and other higher shrinkage filaments it is more important as a filament override. https://www.simplify3d.com/support/print-quality-troubleshooting/dimensional-accuracy/
Please add to PrusaSlicer!

Gefionious, 2022-03-31:
I should note, that it appears there are work-arounds available... I am looking at this: [https://www.thingiverse.com/thing:2484766#Summary]
I think it is very possible to add g-code via the M92 steps per mm correction in a script, perhaps even checking the filament profile. Something to consider.

stepikovo, 2022-05-26:
I'm closing this as a duplicate of #3059

ruredi87, 2022-09-28:
Guys please add this feature to prusaslicer I am begging. I dont know, is it too hard to implement this? Almost all other slicer has this feature and I think it is extremely important feature for the people who prints engineering parts.  I see first topic is opened in 2019. You can take a look superslicer's filament shrinkage ratio. It is probably too simple to add for you guys. PLEASEEE @kubispe1 @hejllukas 
[image]
[image]

johnkray, 2023-01-06:
Chiming in to say I would really like to see this feature as well!

jowgn, 2023-02-15:
Are there any plans to add this feature?

nicoster, 2023-03-13:
A post process script will do?

patrickmcquay, 2023-05-09:
another vote for this, it should be per filament because different filaments shrink differently, as well as have separate percentages for XY plane and Z height, because they shrink differently (IME Z usually doesnt shrink that much).

A nice to have would be something like XY shrinkage scaling based on z height, as a nonlinear function that looks like: 

`thisLayerScale = filamentScaleSetting * zHeight / (zHeight + curveSharpnessConstant)`

because the base of a print is held rigid by being bonded to the build plate, it shrinks less. There is still some shrinkage, but it happens after the print comes off the plate.

DragonDon, 2023-06-19:
5 years....5 years and no setting for 'shrinkage compensation'??  I mean, were there that many other requests that have been around longer?  Please look into this guys!

TjardF, 2023-07-23:
Well nothing to add, need this feature, please add in next Version...
I hate Orca and love how Prusa is used, but with currently only printing abs, asa, pc etc no shrink correction is a no brainer....

MrBlackmidi, 2024-02-08:
Hi, is there an update on about implementing this, also z shrinkage aswell?

lukeskymuh, 2024-02-08:
Aparently the prusa slicer team does not understand the issue. I. Long time ago I got an answer from them that they do not see the need for this simple feature. As other other slicers have now implemented it, they may rethink and change their opinion.

Phil1988, 2024-03-17:
I also link this to this FR, as they are basically all the same:
https://github.com/prusa3d/PrusaSlicer/issues/11183

closed 2026-09-29 by prusaslicer-bot[bot]