prusa3d/PrusaSlicer #3079: Excessive extrusion width

Pun-e, 2019-10-19:
### Version
2.1.0

### Operating system type + version
Any

### 3D printer brand / version + firmware version (if known)
Any

### Behavior
[image]

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
According to some engineering advisors, peak FFF strength is achieved at 400% extrusion width. In my experience this strategy allows for printing in filaments that can otherwise refuse to play nicely such as PolyCarbonate or nylon. 

Up until the most recent update, this profile worked fine, and despite absolutely eating up filament, performed as expected. Now, something has broken this completely standard and viable printing strategy. I don't even care that this warning pops up every time to keep newbies from borking their printers, but there should be a way to permit advanced users to do what they need to do.

Even if there was some way to launch the slicer in "dev" mode or something so I could just use the slicer as I have for years now would be appreciated. Until then I'll just have to step back to 2.0.0.

Here is what the expected result is (2.0.0):
[image]

foxox, 2019-11-29:
I am running into this issue too... I managed to rebuild the source code with these print validation "errors" commented out and that's working for me. I definitely think that these errors should be treated instead as warnings that the user can dismiss. Perhaps in expert mode they should not appear. Or there could be a setting in the Preferences to ignore warnings. If the project maintainers have any feelings about which approach should be taken, I can make a PR for it.

foxox, 2019-11-30:
It looks like this print validation "error" was added in https://github.com/prusa3d/PrusaSlicer/commit/74e592ceaa22ee8793047ccaf26fdadbd46eff5d so I'll tag @bubnikv here

foxox, 2019-11-30:
Here's a proposed fix... at least to get conversation started. https://github.com/foxox/PrusaSlicer/pull/1
I separated errors from warnings in validate()... errors stop slicing, warnings still pop up a message but slicing can happen anyway.

foxox, 2019-12-01:
I looked through the git history some more and found a reference to this issue: https://github.com/prusa3d/PrusaSlicer/issues/2715 which indicates that the extrusion width validations _should_ be a hard error case, but the limits need to be tuned better. I'll make another PR with better limits...

foxox, 2019-12-01:
New PR: https://github.com/prusa3d/PrusaSlicer/pull/3260

Pun-e, 2020-03-02:
@foxox Hey mate, I don't really understand / use github so I'm not sure what to do with these "PR" that you posted up?

As mentioned in #2715 , the limit should be >400%, likely a little larger so you can use measured feedback to ensure that your extrusions are _actually_ that thick.

foxox, 2020-03-03:
@Pun-e The PR is for the project maintainers to consider. It shows the proposed code changes, which they can choose to accept (or discuss, request changes, etc.). I was hoping making the PR would help your issue get a little more traction but here we are a couple of months later and nothing has been done. I don't know what the best way to get the maintainers' attention would be.

foxox, 2020-03-03:
I just cleaned up #3260 (removed some of my unrelated changes) and rebased it on the newer `master`. Maybe that will help it get some love.

closed 2026-09-29 by prusaslicer-bot[bot]