prusa3d/PrusaSlicer #3080: Perimeters not connecting

ioquatix, 2019-10-19:
### Version

2.1.0

### Operating system type + version

Arch Linux

### 3D printer brand / version + firmware version (if known)

Zonestar Reprap

### Behavior

External perimeters are not connecting up and generally this is also reflected when printing.

[image]

See the attached STL.

Jebtrix, 2019-10-19:
You need to attach a zipped 3MF project file for anyone to properly review your problem.

ioquatix, 2019-10-19:
That's weird, I could have sworn I attached an STL file. I tried to attach the 3MF file but I couldn't zip it because the file manager said it was already zipped lol. Okay let me try again.

ioquatix, 2019-10-19:
[grill-project.zip](https://github.com/prusa3d/PrusaSlicer/files/3747258/grill-project.zip)

ioquatix, 2019-10-20:
I actually managed to work around this issue by insetting the filaments, by cutting out a small area around them as they inset into the perimeters. So, I'm happy with my solution, but as it stands the original model should have sliced better than this IMHO. So, let me know if I can provide any more details or help in any other way!

Jebtrix, 2019-10-21:
Thin wall detection has tendency to have this result and is the root cause. The grill mesh feature is such a small extrusion width for a 0.4mm nozzle which is problematic. This could be compensated for in modifiers without Thin Wall detection if PrusaSlicer didn't limit smaller extrusion widths to be used ☹️

ioquatix, 2019-10-21:
Thanks for the detailed response and for taking a look at the mesh.

I was under the impression if I made a cube (or cylinder) roughly the width of my extrusion and height of one layer, it would just basically do a single nominal extrusion the width of the nozzle. I have parameters in my model for the layer hight and nozzle diameter. Should I do something else?

I did work around this issue by insetting the strands into the mesh, but it was a bit of a hack. Sounds like this is kind of what you mean by designing it for the slicer.

So, do you suggest that 0.4mm for my cube width for a single strand is too thin? Not thin enough? Because it should be 0.4mm wide and 0.2mm high.

Jebtrix, 2019-10-21:
Your model mesh strands are 0.2mm wide. The minimum the slicer will allow for width is 0.201mm @ 0.2mm height. So by 0.01mm its too big to fit, that's why it doesn't slice without Thin Wall detection. 

Attached here I used 0.15mm layer height. To make the slicer happy I had to used 0.19mm extrusion width modifier to fit the grill strands itself. I could have also used a weird layer height like 0.189mm which is maximum the slicer would allow to use 0.19mm extrusion width.

[image]

[grill-40mm_tweaked.zip](https://github.com/prusa3d/PrusaSlicer/files/3752003/grill-40mm_tweaked.zip)

Now lets say you wanted 0.20 layer height, that means at least 0.201mm wide geometry to have a chance in PrusaSlicer. Because of inherent floating number precision issues and different STL export resolutions I would make strands 0.21mm to give me some wiggle room. Your profile has extrusions set 0/Auto (roughly at 0.40 ~ 0.45mm range) so you would still need to use a modifier around the grill strands but setting 0.2mm extrusion width would fit this time.

Hopefully this makes some sense. It is definitely more tedious to workaround Thin Wall detection problems.

ioquatix, 2019-10-21:
Thanks so much for explaining all that!

closed 2019-10-21 by ioquatix