prusa3d/PrusaSlicer #3081: Superfluous support structures

emil-e, 2019-10-19:
### Version
2.1.0

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Original Prusa i3 MK3/S

### Behavior
Supports are generated where none are needed, seemingly for no reason:
[image]

There are not even any geometry in some of the affected locations.

#### Project File (.3MF) where problem occurs
[support bug.zip](https://github.com/prusa3d/PrusaSlicer/files/3746416/support.bug.zip)

Jebtrix, 2019-10-19:
Related to #2910, #2416, #1962

I'm not sure what you want supported but just adjusting Print Settings > Support Material > Overhang threshold value can improve results.

Here is using 20 degrees
[image]

emil-e, 2019-10-19:
> Related to #2910, #2416, #1962
> 
> I'm not sure what you want supported but just adjusting Print Settings > Support Material > Overhang threshold value can improve results.
> 
> Here is using 20 degrees
> [image]

Sure, there are many things one can do to improve the results. However, it is still a bug that there is support material that isn't actually supporting anything, is it?

Jebtrix, 2019-10-19:
Yup

nolspeed29, 2020-05-09:
I too am seeing the same behavior...also the lower surface was printed too far from the top of the support layer (more than the .2mm specified), leading the perimeters to detach as it rounded the circle (I printed with the supports on build plate only option).  

[image]

nolspeed29, 2020-05-09:
[open_impeller.stl.zip](https://github.com/prusa3d/PrusaSlicer/files/4604443/open_impeller.stl.zip)
Here is the .stl file.  This was in 2.2.0, with the latest release (and also latest beta) tested.  The original  file was created in Solidworks.  I saw the issue in another file that was created in SW as well.  The file slices fine in Cura.

therooftopprinz, 2021-02-07:
This issue also occurs when using **Version 2.3.0+linux-x64**

Using file https://cdn.thingiverse.com/assets/54/f5/04/8e/cd/xyzCalibration_cube.stl

![Shitty](data:image/png;base64,UEsDBBQAAAAIAM01R1KqczBtvZ0AALOdAAArABwAU2NyZWVuc2hvdCBmcm9tIDIwMjEtMDItMDcgMTMtNDEtNTctbWluLnBuZ1VUCQADsX4fYIR+H2B1eAsAAQToAwAABOgDAAAAdYCKf4lQTkcNChoKAAAADUlIRFIAAAKiAAACaQQDAAAAhABNtwAAADN0RVh0Q3JlYXRpb24gVGltZQBTdW5kYXksIDA3IEZlYnJ1YXJ5LCAyMDIxIDAxOjQyOjAyIFBNCK2/WAAAABl0RVh0U29mdHdhcmUAZ25vbWUtc2NyZWVuc2hvdO8Dvz4AAAAwUExURbFWJhYWFiAgICssK3o7Gzw8O/b29gF8AQCsAJ05JgFVRa+bNYt9KlpYWpSUkMXHxBgkOxIAAJzaSURBVHjaxL1NiGTZlSZogIMwMhHY3eRQu3vfQIFWU+/NQ4ygwUuy6ChjQj3T/crQtMC3UUtHmVJaUCAlMEDEZhgDDAqD3JSEaEioJoDaFBEMAgKSIFxiNkahRpFoVQ2MwmPntYuYd8/fPee+9+wvzL1MVfnjEeaRfuy88/ud7xut29dqusbXdIZ/X834C6v41wevnj27al/P+G/Pr65my9FoPZ2Olst1+0+jj56O9Gvd/t9qevVKvSH+y/Or316/u77+Q/qK/PIU35Jek4kLRRVfRQhuIq9Q1fOLi4vNfF5VIfBXnQtVVdfz+bxpqqoMwU8m3rWviQ/x5dtX+4UQQoWv9NZJCEUV39m+t31n+qPgezb05fbt8QvxG7sQygK/qYvfvcTvGH+Pm/jReqtJ4RdmD16xBV5dvYh2aP+GFpgpK45HYttV+4tjsOjz5/RJvGj/1v7L767fX19fyfdrv/LgFRh2im+R1z22RcAfJv38RQU/PRjUy08fiqImszT6F3zAz4VMMAnxdxZFKEMIDr6rSyatW7NM1Fv5j4rf0aGN4bezCeO3cPDfWBTwLyGE1ipoynWPKflr6EXJrq0BXkSDvpiej3pe41XrwOPZlN5gPPyb1kff4Ld4kHy+/X7tZ9O+Rb28m5BNHf1nRzeb4M+DfhPACx399EVrlA0YtFSWdtGn67pq4AOIjhrMK37Fx+8Jzl2Bn5FB4a3pw/NgazfhpwEcPZDtJ/Hjgf/K0Xo94KVTZdHs6W09axqNtu636OgshoSzFZvsufok2KLkuxINrvAtyaB/qX60+J/aukRZgn2jR4Toc54914HllO+KQcHPGvIzcF34rtGGRUF+5tDFS4wFHl7x0/Q+wHe8QB/19CeBRXv+pAkFgJHYbdhLp+iVaJ8HFPZaA6xW01mvReNDv55RGI1h4rm87/rt9fWfku9iFImfEL4lvVYqoPkYzxoKkIGeM4+G8Wx5fEZrMIsTB3WODV2z6zr8AEqwC/hZcMlx6bvj80ymE2eMnyH8OQ2HF/3ZxfDSNMmi60EvfZAnkvZvs/Vo/LJ1KXDSp5lFZ62hv7Rh9ArD72+jRWNiepV8F375xfkyvkW95D9WuV8DjyU4GeWZ6Glg5ug76HeFPLnxt5KPbtr3FhJePYROcr8SI/UEYouH71kEehokjcX4iYkufsfko5OJDvAxSIyerge9lL4Q42FyMnK86JytAdhHx8txyvhjrAPooY+Z5zkmtKs2McVUDz5qfZdKB5WZ0scPP/yGc3GYJKtUMRfXFOVShHQO/5Wf0UYlsvg8Y9jFb1orn6a8z2nHBzYv+TZmeoeBHOKusqijsDt69uzp9gd/Oo2WQWeKD+nz6HgvVhBGe576NuOfTcGBH0BtYMNvDKPX0TmzYuzFlHxe+WiWcjevL9AsMQVAkkCrqKwfQx8/uCUVCWj3msobfnTB/dB1TcXkJBtVlccHHesu/LRKePo9fpS2BKO6DiyqbNpTj06n6GTy0GOts4Tnuj8zpdopOSen+vahfxNt+Jyc80Xy+bEx6F+mp14e+tah4EeC5BLtVlF6j0XPZNJTMIFFQyjbF/6rw3QNpW6jEhbbU9dRumAIZVVw6kKbBgo2yknhvxMs+uwZl6Wz3Et/+O8urtKPjnZ4hcVTLHZ6LXrW1qVfrqfsnMp+VxhGX3FgFd990b5laZ7673Nlgj89B67oFlSeevw6ua5KEfhc0s9PNY0US2CS+D2LWAPpWh/DKJcGptYPEDoowJCbYu3mk9kn+MeSRdGm0+9fTPNq6ocXPZ3PCyw429foWdeiUwmjz837YmJ69w7C6FXmu/EtZ1PzqqQup66kqVIzFJ/7mK4huXR9VOVoMYvnsMjfs6xKCQ34NkhLVd1abl7pb6kLBv4AvMPAWhQVNQveY4X/TF7tT1S9fv37ToFqWhvVg0L1GC2IWUlsO57pp56cGt8HickEVk5M7VN/vs4sGivzGlM5xLDS+BomEXlwJ6nXmaSCsVQmdVTeYO7mFzsh+i88DzX8JvPO9JSUynchvEixT18ePXuebPrb16+TSTlPrZKvYVHKTXj0UZuZMOOfUcdknBr/5Zu37yCMPst8F8Ko9dF7qgEME7QXlPpBCprW3Sr4cUqdIeJvgoY06/y16+IbnE7wBZu4Srk+eW98GtppAna41qCpkHJk0fbHSzZ9c9Hx0lWqJ81T/3F00fOBjmk8ljD6XCU01TE96Ju7qNcPVGGODTjFvwp+iAYf2RAKfHRjxURtKjbedWoec8+lKJHaALJiIeVoCX9ocF47N3++3quPDsY2JuziU/9c23TzWqf+1YNXyjm5Y2pdCnz0fCDVfznVHdNz6o+u4pjkD2Y8kHx+ZVL9tLFDCof5iJxiw84Sv8y9esF9OmRmaP11Lo6e21pls4Gf3zs2KZUM8UGXIZebUKgtKRyUqdR31PaCj5ZUKTeNhF2Oo9pPX7NJZZSXjd5exEj5cjqd9nahMEVaZx3TFY7y3mIYxbpWGbbtmEazLDPVErjQ9+IPQVUkP9BOHlsqOmmeQg8vZbHUKKipFTSjfiKVqWrI0m9X2bHAFKTrTyxMqR8t+a0pMxk/VYOnBz2DJ6gewafWcRQ60hk/dkxtXpKOSSV2TEypauJvC6VD5qPTImUdD/00NJpQgbYFU0OVEcZXeswbquJ5wKQHTT5wLGgkjyeDxie39bPSDk2T+ePgM7iAf5iDpszhH1NRGpOPYvQqGVL56YtZz+Dp1dWDVD2O15mPLsdo27PYTr1cZcXTM+6Y3lz1jK/bD+bjpTXoD3iSG6g2gYqloAzB+X3Cj18hLh1cSvtOMhmaNXWYYA+MI/zWWvsoNE/yLbMJU4HBG0YuoeQ4n576fpv+YzZ4yqvHL6cDT73qmF7xwOoq1feqY5JvGx/6ceaiUyzDo4NKux7TdZ3qIugQpQKnrIM510s4rGjEBD93jHxcg8XE470LMqdP404196BRXirjMb7UVAzg5FbFCnzqn/fb9CwadDbV1eNVqh5xg9GTmeLkVHVMr3ha8gzD6J9ST/tMhq3th7DMnvq/xAoaEk3rRmhb8Ci0ULYqkb0G5VwPYUIcTfsfpZ30ai0EVWqpG0/+LDZzO8HmwQu3D96pIiykCn/QplPjZA9S9fhyyEdnumN6oItZtWPKfL6b6qc/wNAZvUie1YkeMAW0jOcU0fQM2KjX2UhVG7/fJHWp+PRCI1Xogh+3LFVdtZ9FKvbjME/PGCTqglUxvJQlZ6YBP32BkfPFle58Xsw4jELPZJd245kMniT8UhXLiel58l3y+ZftyCp/6v9CcgcX5lR045SdZiU4oQOPqut89ZbeG4ejLpukoF9ituHFC2cYh6VuxaHcc0OAD8NmA/OZVGxQ3ovLF9WF9tv0ebfzmS7TdJhXn5Lxz2aqBbVVJw+enplh6yv4hEbTaSeO4l6HshFvJLmt54EwTuDjP8fpUFHaqEZPvcxVPcaCwuw1vC31i9B9cU0Pk5RaOq9u91Q3I228fpuyk8W/ytaOtmydp365Wqun/oUxLITRFJBfpa1dbLI6Tz3OcyY0bI8lKMzjvOehRIHrtiKVR/h3NzENeUV5jAsm/Pl5xhImnuMLh1Qsh0Iai6YKzEm9UJl5lwwW51VrhJ02JSeTBK0e+vPe+n6VreqxKP3t9fu3umN6zpVAKh3sNA9XSCHlXNqaT7we8eEyOabuiURZaRTJAKrAcfRF5WdOueg8XxVwCNCv2KYW/Fs87VPQR5tmNNpp01finJSgp2NZW3YsOsa8TYXr1SuVgNSO6UUaX8OwNX1EZj6K6UdMB7nYYeMZM0eDLTXuNciKOByuit5nl/wMHa2mbTJ2leikZj/vrafLij570Xqba/2KDLHNpldXWfWo15ZPO2OS1qovdKqXMPo7W99Tt88dUyeOUsHEa526oqoa9mdgt/icUb0IxaVUPQ2N7LDXkhqpVPYoUqmP8ZpKzT5oCU2hG1tfBbM3hMIWGlUxxaBNn1+ZwdMVrOpf8ujNQEl4Vb+WVb1+xr+xPWiqBOJT/7JjUrKKI8BGif/IkBt8RC+opwbfwyVp1j2pBpzdrEyYCqwoPWQ3j49DI6tAb/dHFCTUl7G+QJumsk6514BN713BSE4NnsZp8jbLlna4qk+Dp+dmVf+WEU+m238xyxBPOB+VRbnkW4AYQOHoU+TCzZtzwVNyqlIqlimJU8NpC1uYOKc7/6IDioISDItaSnC0u5b0Dmt7J2uCNsnutCnGQUnQkJlnA5lpeo4dU9rFP9eDpz/pupae/Wnr4x+vey1Ku89oMKccDdtybvzLoDKySz6qs34C3OBwlJ9057nx521yWVYVNqyOpgmuoMarAShA+pYx7lxQozbhYi+Aa+22qRk8LRNC6XyoY+qu+H+rV/XPsilBx6DKRzHSxYdSQzzwuS1gKqocjc3ckHurcXOdZnY4wKYSrDVXXVHWdzBYSsMEB8NXxjnZXQH2aa9jdqw8YdKcw6d+Dz9NxSWsLcd9T30Mo7yq17t4zPu8Y3puuv3nCE8ZzzomxYKpgKGdB5OmaVDVb8bSJh0oH1XjX+OipAwybdZrjVJPkJzqdePvKotQViHYeCF+31QAM8PGQ0LgLpumyZtZW44teFQNnq4yV/ymDaNZxyTl7ahr0Om8mTO0y0HnrGobHDzhQCRwA4gAnZAhmLztR9HRHKHP8KvQUjZm9wkDkUrFYs9vmRiQWpr3eU5XqtvZatNXVxromdaWKSuBbVe4d09YJ9qiPk8dU97tR4sue556TsyAcIKinjcisZDyAWFIqujBpE/YJQF2MsQm8H4enB5t6niUhX7vcnhItp+WNOZSR9A0DcZXeafyzq1+emWAnuKjs55V/ZfTqXFOtN9vB3dMGcxRUDoRCAIPvTSJFAsRLUIPmnpyaWap8i7 [... cut ...] g93hRa21oyI4Wp0TguqmHpvBmN0S4ZMdJHpkmjgg1Gjr+sVe5xpEsVXpu8T5Vo6v3GvXDjT2+udVvWoSmSK5RlRJyE4UoHKgNAP5gSa+DJZgDrNRh3pOM3qrDFFzU/Q5BCaQZSxeV//O5cvIU7fuzhd3ssUeNWsNECQEVNgIqADd9YYLnVDC//YtJsMOLrJeYqTLJVokAZOFj+27wqfeYg5Dkjz1Rzp35ozvfEh2vlUV8nClkbJJFFIhNFCEdXI0QvAz6KtBPCD3Y0UPQm2DgVXI/KjAfhc6Pn/BZq2TmRcIT3YAAAAAElFTkSuQmCCUEsBAh4DFAAAAAgAzTVHUqpzMG29nQAAs50AACsAGAAAAAAAAAAAAKSBAAAAAFNjcmVlbnNob3QgZnJvbSAyMDIxLTAyLTA3IDEzLTQxLTU3LW1pbi5wbmdVVAUAA7F+H2B1eAsAAQToAwAABOgDAABQSwUGAAAAAAEAAQBxAAAAIp4AAAAA)

bubnikv, 2021-02-19:
Fixed with fb7b9950501d01164955b7dd29f163005552234e
[image]

closed 2021-02-19 by bubnikv