prusa3d/PrusaSlicer #3082: Object becomes shifted outside printable area during slicing

DrLex0, 2019-10-19:
### Version
Latest master (f6057c1b4f06f02882b4c28b2054d64380c1874a), own build

### Operating system type + version
OS X 10.14.6

### 3D printer brand / version + firmware version (if known)
FlashForge Creator Pro, but the problem happens way before sending anything to the printer

### Behavior
* When I slice something, the sliced object is moved by half of the bed width to the left and half of the bed depth to the front, usually making the print impossible.
* _Steps needed to reproduce the problem_
  * Open the attached project file and try to slice it.

* _Expected Results_
Everything is sliced where I positioned it
* _Actual Results_
  * The sliced object is not where it was placed and is now outside the print area. I thought this had something to do with the fact that the bed origin in my profiles is at the center of the bed instead of at a corner, but I get the same result when I set the origin to (0, 0), so I would expect this to happen for everyone.

[image]

[image]

These screenshots also illustrate issue #2555 which has still not been fixed…

#### Project File (.3MF) where problem occurs
[Box.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3747079/Box.3mf.zip)

DrLex0, 2019-10-19:
I jumped back a few days ago and this bug was not present at commit 62b7892fa112a5ee7fef62aec5a48276419685ea, so it was introduced somewhere after that…

enricoturri1966, 2019-10-20:
@DrLex0
This issue is caused by line:
"G92 X118 Y72.5 Z0 E0 B0; set (rough) reference point (also set E and B to make GPX happy). This will be overridden by the M132 below but ensures correct visualisation in some programs."
of start G-code, which is now processed by GCodeAnalyzer after commit: https://github.com/prusa3d/PrusaSlicer/commit/0285e8db2d6455618e6aa1f49780cec7d6090379
What's the purpose to use values for X and Y different from zero ?

DrLex0, 2019-10-20:
@enricoturri1966 The purpose of the G92 command is to tell the printer, as well as any visualisation or G-code conversion programs, that the extruder must be assumed to be at the given coordinates relative to the origin at that moment in the G-code.
This particular printer has its origin point at the _middle_ of the bed but it homes to the _far right_ corner, therefore given the geometry of the bed and some margins, the extruder will really be near X=118 and Y=72.5 at that moment because we have just homed and have not moved yet. The true, calibrated home coordinates are subsequently loaded from EEPROM with the M132 command. Because the slicing program cannot and should not know the EEPROM values, it has to ignore the M132 command and rely on the G92 arguments instead to produce a correct visualisation.

You seem to assume that passing any X and Y arguments different from zero in the G92 command is never done. I hope the above explanation illustrates that this assumption is incorrect.
The previous interpretation of the G92 command was correct, and produced accurate previews in both Repetier-Host and PrusaSlicer before this commit. The sliced preview should look like this:

[image]

enricoturri1966, 2019-10-21:
@DrLex0
Could you please check if commit: https://github.com/prusa3d/PrusaSlicer/commit/2d610f9b84df1be44682e2fd1a94f05592f39841 helps in fixing this issue ?

DrLex0, 2019-10-21:
@enricoturri1966 Yes, with this commit it looks correct.

bubnikv, 2019-12-17:
Fixed, it will be part of the PrusaSlicer 2.2.
Closing.

closed 2019-12-17 by bubnikv