prusa3d/PrusaSlicer #3083: Huge ratio of unneeded supports despite support blocker

Chrismettal, 2019-10-20:
### Version
2.1.0+win64 201909160915 

### Operating system type + version
WIndows 10 Pro

### 3D printer brand / version + firmware version (if known)
Stock MK3S

### Behavior
I have the following feature in a part i am designing right now:
[image]
The chamfer is intended to only make supports neccessary in the uppermost part between the parallel features, while not supporting anything below the chamfer.

Additionaly i added the following support blockers and enforcers:
[image]
I was hoping the blockers would make supports at this height absolutely impossible.

What actually comes out sliced is this:
[image]
The part is not directly supported on the 45° slope, which is good, but the upper part feels the need to get supported "from the outside" and generates supports through the whole height of the model. 
I expected the blockers to prevent supports at these heights of the model.
In the actual part i am designing the full height of the model is much greater and there are a lot of occurences of this feature. The supports alone even with a 0.6mm nozzle add up to around 4hrs of unneccesary support printing.
In my actual model they don't even come in from the side but rather just support the outermost part of the chamfer, effectively doing nothing but waste time and filament.

Maybe i am misinterpreting the actual function of the blockers. I feel they only mark SURFACES that need support, while i was hoping they would stop support generation in thie AREA i marked. Is that right?

#### Project File (.3MF) where problem occurs
[Testpart.zip](https://github.com/prusa3d/PrusaSlicer/files/3747637/Testpart.zip)

mroek, 2019-10-20:
That is quite unfortunate. I played around with your test file, but I found no workaround for it (but I'm not that experienced with PS, so it may still be possible).

One of the things I tried was to use a height range modifier. However, even if you can choose support material as a setting for a height range modifier, it doesn't seem to actually work. Not sure why, but when clicking under the height range modifier to add settings for support, nothing happens.

Jebtrix, 2019-10-20:
[image]

[Testpart_tweaked.zip](https://github.com/prusa3d/PrusaSlicer/files/3748196/Testpart_tweaked.zip)

Jebtrix, 2019-10-20:
Just a fyi I wouldn't print this part in that orientation. Laying it down would be a lot less print time.

[image]

Chrismettal, 2019-10-20:
Wooah when i'm back at the pc i gotta check out how you did that. Thank you!

I know laying down is better. This is just a demo piece of the specific feature the slicer was having problems with. My actual model is a lot bigger and cannot be oriented differently while having a ton of instances of that feature, intended to save on supports.

Jebtrix, 2019-10-20:
It's worth noting this workaround is very _picky_. That same setup rotated different ways:

## Rotated 90
[image]

## Rotated 45
[image]

termlimit, 2019-12-06:
Adding to this issue, the attached 3mf is displaying the same issue. I have tried "For support enforcers only" and "Everywhere" and each has its own variation on the problem.

I am including the best version I could come up with for the two support options listed above.

[image]

[image]

[FrameSlice_Front_v2_support_options_3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3934287/FrameSlice_Front_v2_support_options_3mf.zip)

bubnikv, 2021-06-03:
Implemented with "snug" supports

[image]

[image]

It will be contained in PrusaSlicer 2.4.0-alpha1.

closed 2021-06-03 by bubnikv