prusa3d/PrusaSlicer #3087: PrusaSlicer 2.1 and 2.0 Ignoring Settings.

Design-Create-Innovate, 2019-10-20:
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

Design-Create-Innovate, 2019-10-20:
Ok I just noticed something, when I changed the filament profile in the main tab it did not reciprocate this change on the right hand expert settings option when viewing the platter. When making sure both are the same selected profile the problem goes away. I just didn't think I would have to manually change both...

JohnnyDeer, 2020-03-17:
Hi, can you please try to describe steps you did before exporting gcode? Which parameters you changed? Did you switched profile after changing parametres, or before?

rtyr, 2020-06-18:
@Design-Create-Innovate Is this issue still valid? Do you experience this problem in current PrusaSlicer 2.2.0? If yes, please provide your project 3MF file.

Design-Create-Innovate, 2020-06-18:
No I figured out it was just a lack of understanding. I would change settings in the main print settings tab but on the right hand side of the screen I realized it was on a different profile and that’s why it never changed. 

I would however like to see better support for dual extrusion with multiple print heads if possible.

rtyr, 2020-06-18:
OK, so we can close this.

I am not sure what do you mean by better support for dual extrusion. You can open separate issue, if you have some suggestions. But I suggest to check existing issues first.

closed 2020-06-18 by rtyr