prusa3d/PrusaSlicer #3088: Enhancement: Default "Print settings" and/or default "Filament" per "Printer"

quiccsilvver, 2019-10-20:
### Version
2.1.0+win64

### Operating system type + version
Windows 10 / any

### 3D printer brand / version + firmware version (if known)
Prusa MK3S, Prusa MK2.5S, etc.

### Behavior
* When switching the "Printer" the default setting for "Print settings" is always "0.15mm Quality MK3" or similar.

### Enhancement
* The ability to specify a default "Print settings" profile per "Printer".
* Along the same line, the ability to specify a default "Filament" profile per "Printer". If not set it falls back to the filament used by the last printer.

Jebtrix, 2019-10-20:
I do have some profiles designed to be used in combination of others myself. What I've been doing is just manually editing my printer profile ini files. Exposing it in UI would be useful indeed.
```
...
default_filament_profile = "Prusament PLA"
default_print_profile =0.15mm Quality MK3
...
```

Ali229, 2022-01-23:
> I do have some profiles designed to be used in combination of others myself. What I've been doing is just manually editing my printer profile ini files. Exposing it in UI would be useful indeed.
> 
> C:\Users\<username>\AppData\Roaming\PrusaSlicer\printer
> 
> ```
> ...
> default_filament_profile = "Prusament PLA"
> default_print_profile = 0.15mm Quality MK3
> ...
> ```

As of PrusaSlicer 2.4 this has no effect for me.

Ali229, 2022-08-06:
This is really needed, so annoying that when changing printer it selects the default profile instead of my profile

Ali229, 2022-09-15:
Any update on this? Super annoying to select print & filament settings whenever I change printer.

DragonFinlay, 2022-09-15:
> Any update on this? Super annoying to select print & filament settings whenever I change printer.

We're currently focusing on other tasks as this is more of a quality-of-life improvement, but I'll forward it to the team once more and give an update as soon as I know more.

nirurin, 2022-09-20:
This used to work, why are the default_filament_profile settings in the .ini files now being ignored?

DragonFinlay, 2022-09-21:
Just to keep you up to date, the initial description of this issue isn't valid anymore, the logic was changed about 2 years ago, with PrusaSlicer 2.3.0-Alpha1. Here's the info from the changelog:

```
When switching the Printer profile, the active Print and Filament profiles may no more be compatible with the new Printer profile.
The old PrusaSlicer selected a Printer and Filament profiles marked as "default" in the activated Printer profile.
The new PrusaSlicer selects the new Print profile with the following priority:

    1) Compatible Print profile with the same logical name aka alias.
    2) Prefered Print profile with the same layer height.
    3) Compatible Print profile with the same layer height.
    4) Prefered Print profile.
    5) Any compatible profile.

Thus when switching the printer from "Prusa i3 MK3" with the "0.2mm SPEED" Print profile active to "Prusa i3 MK2", 
"0.20mm 10mms Linear Advance" Print profile is newly activated, while the previous PrusaSlicer would select the default, 
which is "0.15mm OPTIMAL". The same logic applies to the selection of Filament profile, while the symbolic "filament_type"
parameter is matched instead of the layer height.
```

We are looking into this and the logic might change once more at some point, but for now we can't say when.

Ne0nx3r0, 2022-10-31:
Very confusing given that when you look at the Dependencies section of the Printer Settings it shows the default print profile that is then ignored.

stefanbayer1, 2023-02-04:
Is anyone at Prusa using two different printers and having this problem? I believe not, changing printers and always selecting the Filament and print settings again is beyond annnoying. Why is no one working on this?
Currently i have a MK3S and a MINI- i am even considering selling the MINI just because switching in the Slicer more then 100 times a day is just to cumbersome.
This needs to really be fixed- please!

matthewOrichardson, 2023-04-17:
+1

Please expose the option to change the default profiles. Using multiple printers is made far more difficult when it defaults to a profile that you would never use.

Misfi, 2023-06-23:
+1, it's fairly annoying when switching between MK3/MK4/MINI profiles and having to re-select your custom print settings every time.

BeardOfSpiders, 2023-08-16:
It's amazing to me that it's been nearly 4 years since this issue was brought up, and this hasn't been implemented. I've tried the .ini editing workaround, but it doesn't seem to stick 100% of the time.

stefanbayer1, 2023-08-16:
Yes amazing to me as well! Seems the Prusa software engineers are not using their product- otherwise this would have been definitely fixed.

dssamusaran, 2024-01-21:
We went online searching how to assign a default printing profile per printer. We were quite certain we were being stupid and were just not finding where to set that up.

So yep we're amazed as well this is not a feature. And moreover the way the slicer auto selects a profile seems to works completely randomly.

We swap nozzle multiple times a day so we change 'printer' in the slicer as well.
And as we often do time-lapses and such we do have custom print settings with little modifications like verbose mode, retraction, etc...
Same for the materials. All is custom with a multimaterial single nozzle system.

Each time we change the printer the slicer comes back to the default prusa print and material profile that we would never use. This leads to many mistakes if we forget to swap back and failed prints.

Please look into adding that, or at the very least take into account the .ini file default print and material profile...

(I do say 'we' because we're multiple people working with the printer, not because I'm big headed :D )

neophyl, 2024-01-21:
Just use saved projects as a workaround like many of us have been doing for years.

A project will save the profiles in use as well as any modifications.  I use 3 printers and I have MK3_default.3mf,  CR10_default.3mf and a Neptune3_default.3mf.  I just load in the appropriate project and the profiles I use are all there.  

Given that you can also import all the settings from a given saved project into the current project you can build up quite a collection of presets if you want to go that route.  

Just because its not nicely into the UI doesn't mean there aren't ways to leverage the way things do work to achieve the same ends.  Just think of a project as a preconfigured workspace.  

One other thing we do is create small non printing objects saved in these project workspaces and attached to the objects are commonly used modifiers.  You can then copy a modifier to an object, that way you don't have to recreate those each time either.  You just have to think outside the box sometimes.

dssamusaran, 2024-01-21:
Thanks for the input. I have to say I didn't think about that!

There's a lot of outside off the box'ing in our system, to use for example color swap to go from Pla to tpu and such easily.

Did not think about project for each preset to have a starting point!
Only issue would be when we change nozzle size upon slicing to improve features or time. This would need either changing printer and then profile, or loading the saved preset.

But thanks I see there's something to do there to get things a bit easier in the meantime. 

Still I believe the software should recognize and use the 'prefered_...' stored in the .ini

democat3457, 2024-10-15:
+1 on this. Would be helpful if defaults also applied system-wide. Currently the filament defaults to ABS for some reason (when we use PLA), which trips up users who aren't paying attention.

noffma11, 2024-11-26:
+1 
Having this feature would be helpful as I have new users needing to slice files constantly in my line of work. If I could set some defaults based on the machines I have would help to eliminate the issues I'm currently having with improper slicing from less experienced users.

Misfi, 2026-09-07:
@quiccsilvver you might want to re-create this as a Discussion. All issues unrelated to bugs will be closed.