prusa3d/PrusaSlicer #3089: Feature Request: Link STL file in .3mf instead of embedding

michal0A, 2019-10-20:
It would be beneficial to be able to link STL files to a project (3mf or a new type might be needed for this) instead of embedding them. This would allow for the much faster and less error-prone process while going through design iterations because the only thing required for print another iteration would be to update the STL in a folder and generate G-code again. Otherwise, we need to remove each part from a project, add a new version back and set up every feature from scratch (orientation, scaling, extruder (in case of a multi-material object), supports, and so on, which often leads to the inconsistency between iterations setup caused by human mistake.

michal0A, 2019-10-20:
I know there is a Reload from Disk option, but it doesn't work as expected, it seems it uses absolute paths so this way I can't, for example, copy&paste project file and keep older revisions in folder structure behind.

Jebtrix, 2019-10-20:
Duplicate of #3073, #2526

I'm looking forward to this feature myself.

Jebtrix, 2019-10-20:
Actually this is already being implemented and working in latest git. I didn't notice because I use AMF files and that currently has issues.

michal0A, 2019-10-21:
Thanks for linking your issues and the information provided. We will see once the next release it out.

bubnikv, 2020-03-15:
PrusaSlicer 2.2.0-rc3 improves the reload from disk feature. I think it should satisfy your needs.
Would you please test that?

rtyr, 2021-02-03:
Implemented. Closing.

closed 2021-02-03 by rtyr