prusa3d/PrusaSlicer #3091: the application/mouse/whole desktop freezes completely and frequently

joysfera, 2019-10-21:
### Version
PrusaSlicer-2.1.0+linux64-201909160925.AppImage 

### Operating system type + version
Ubuntu 18.04 with nvidia gfx and in gnome-flashback mode

### Behavior
Mouse and the whole GNOME desktop freezes frequently for several seconds as can be seen in the linked video recording of my destop.

As soon as I launched the PrusaSlicer I started moving the mouse pointer in circles. Whenever the mouse pointer stops in the recording then the whole desktop got freezed - you can see it for example on the desktop clock in the upper right corner.

This makes using PrusaSlicer totally impossible, unfortunately.

It started happening in version 2.0.0 sometime in September 2019 and I would swear it was directly related to PrusaSlicer downloading some of its updated config settings (you know - as it asks for downloading updated stuff as soon as you launch it). I even tried returning to previous backup of the config stuff but it didn't help. Later I upgraded to version 2.1.0 in the hope it would get fixed but nope.

Please note PrusaSlicer/Slic3rPE has been working great since first AppImage you released, till the September 2019.

The video is here: https://youtu.be/MYM6-f_S1b4

Jebtrix, 2019-10-21:
Have you tried:
* Completely removing/renaming profile folder and let it build a new default setup?
* Run Ubuntu 19.10 Live DVD (since it finally includes Nvidia drivers by default) then download and run PS appimage right from the live environment? 
  * At least see if it can be narrowed down to v18.04

joysfera, 2019-10-21:
Just tried removing ~/.PrusaSlicer - the freezing now occurs in the configuration wizard, almost immediately.

Will give Ubuntu 19.10 try in a moment, will report back.

antonsperling, 2019-11-14:
I am using Ubuntu 19.10 and can confirm the freeze of whole system while working with prusaslicer.
Maybe i can send you some kind of log files? dunno which would help...

Hardware:
Lenovo ThinkPad T480
Intel® Core™ i7-8550U CPU @ 1.80GHz × 8 
GeForce MX150/PCIe/SSE2
Software:
Ubuntu 19.10 with latest patches, kernel
GNOME 3.34.1

lukasmatena, 2019-11-15:
@antonsperling @joysfera
You both run Ubuntu and have nVidia graphics card, which we already had issues with (such as https://github.com/prusa3d/PrusaSlicer/issues/2939). An OpenGL driver could be the cause, @enricoturri1966 might know what to do.

You can run PrusaSlicer from the console as `./prusa-slicer --loglevel=9` to get a detailed log and see what happens before and after the freeze. But the actions may not be logged, maybe this won't help at all.

joysfera, 2019-11-15:
`loglevel=9` doesn't output anything before/when/after the GUI freezes.
I still wonder how it can freeze the GNOME when fileselector pops up. One would think that at that time the PrusaSlicer does not do anything and just waits for the fileselector to return selected files. Yet it manages to freeze the whole system. No other application has ever done that to me. BTW, Chrome also uses OpenGL, i believe, yet it doesn't cause any issues.

antonsperling, 2019-11-19:
again, i can confirm, that loglevel 9 did not print anything noteable.
But there are a lot of error messages in log. Here are just the first some lines immediately after startup:

> $ ./PrusaSlicer_latest.AppImage --loglevel=9
Gtk-Message: 13:51:22.978: Failed to load module "canberra-gtk-module"
[2019-11-19 13:51:23.004528] [0x00007f784e618400] [info]    Checking if indices need to be installed from resources...
[2019-11-19 13:51:23.030531] [0x00007f784e618400] [trace]   translation_language provided by PrusaSlicer.ini: en_US
[2019-11-19 13:51:23.030625] [0x00007f784e618400] [trace]   System language detected (user locales and such): de_DE
[2019-11-19 13:51:23.030639] [0x00007f784e618400] [trace]   Switching wxLocales to en_US
[2019-11-19 13:51:23.380361] [0x00007f784e618400] [info]    Undo / Redo snapshot taken: New Project, Undo / Redo stack memory: 0MB Resident memory: 68MB; Shared memory: 51MB; Private memory: 17MB; Peak memory usage: 68MB
(prusa-slicer:20775): Gtk-CRITICAL **: 13:51:24.403: IA__gtk_tree_view_scroll_to_cell: assertion 'tree_view->priv->tree != NULL' failed
(prusa-slicer:20775): Gtk-CRITICAL **: 13:51:24.810: IA__gtk_tree_view_scroll_to_cell: assertion 'tree_view->priv->tree != NULL' failed
(prusa-slicer:20775): Gtk-CRITICAL **: 13:51:24.811: IA__gtk_tree_view_scroll_to_cell: assertion 'tree_view->priv->tree != NULL' failed
(prusa-slicer:20775): Gtk-CRITICAL **: 13:51:24.811: IA__gtk_tree_view_scroll_to_cell: assertion 'tree_view->priv->tree != NULL' failed
(prusa-slicer:20775): Gtk-CRITICAL **: 13:51:24.811: IA__gtk_tree_view_scroll_to_cell: assertion 'tree_view->priv->tree != NULL' failed
(prusa-slicer:20775): Gtk-CRITICAL **: 13:51:24.811: IA__gtk_tree_view_scroll_to_cell: assertion 'tree_view->priv->tree != NULL' failed
(prusa-slicer:20775): Gtk-CRITICAL **: 13:51:24.811: IA__gtk_tree_view_scroll_to_cell: assertion 'tree_view->priv->tree != NULL' failed
(prusa-slicer:20775): Gtk-CRITICAL **: 13:51:24.811: IA__gtk_tree_view_scroll_to_cell: assertion 'tree_view->priv->tree != NULL' failed
(prusa-slicer:20775): Gtk-CRITICAL **: 13:51:24.811: IA__gtk_tree_view_scroll_to_cell: assertion 'tree_view->priv->tree != NULL' failed
(prusa-slicer:20775): Gtk-CRITICAL **: 13:51:24.811: IA__gtk_tree_view_scroll_to_cell: assertion 'tree_view->priv->tree != NULL' failed
(prusa-slicer:20775): Gtk-CRITICAL **: 13:51:24.811: IA__gtk_tree_view_scroll_to_cell: assertion 'tree_view->priv->tree != NULL' failed
(prusa-slicer:20775): Gtk-CRITICAL **: 13:51:24.811: IA__gtk_tree_view_scroll_to_cell: assertion 'tree_view->priv->tree != NULL' failed
(prusa-slicer:20775): Gtk-CRITICAL **: 13:51:24.811: IA__gtk_tree_view_scroll_to_cell: assertion 'tree_view->priv->tree != NULL' failed
(prusa-slicer:20775): Gtk-CRITICAL **: 13:51:24.811: IA__gtk_tree_view_scroll_to_cell: assertion 'tree_view->priv->tree != NULL' failed
[2019-11-19 13:51:24.812048] [0x00007f784e618400] [info]    Checking for cached configuration updates...
[2019-11-19 13:51:24.815268] [0x00007f784e618400] [debug]   Vendor: Prusa Research, version installed: 1.0.4, version cached: 1.0.4
[2019-11-19 13:51:24.815289] [0x00007f784e618400] [info]    No configuration updates available.
(prusa-slicer:20775): Gtk-CRITICAL **: 13:51:24.830: IA__gtk_tree_view_scroll_to_cell: assertion 'tree_view->priv->tree != NULL' failed
(prusa-slicer:20775): Gtk-CRITICAL **: 13:51:24.830: IA__gtk_tree_view_scroll_to_cell: assertion 'tree_view->priv->tree != NULL' failed

rtyr, 2021-02-03:
Is this issue still valid? Does it happen with PS2.3.0?

joysfera, 2021-02-03:
No issue for me anymore as I got rid of NVIDIA gfx.

closed 2023-02-08 by lukasmatena