prusa3d/PrusaSlicer #3092: Can't add multiple instances when Auto Center is turned on

tg73, 2019-10-21:
### Version
Version:   2.1.0+win64
Build:     PrusaSlicer-2.1.0+win64-201909160915

### Operating system type + version
Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 18362), 64-bit edition

### 3D printer brand / version + firmware version (if known)
MK3S/MMU2S - but not relevant

### Behavior

With auto center turned OFF, you can select a part and then click the `+` button or press the `+` key many times to create many copies/instances of the part.

When auto center is turned ON, after clicking `+` the first time, the parts are recentred, but then the active part is automatically deselected so clicking `+` again does nothing. I suggest that auto centre should not change the selected part(s) during its operation.

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
Any. Just add any one part and try to create lots of instances with auto centre enabled.

Jebtrix, 2019-10-21:
This is already fixed in git master.

rtyr, 2021-02-04:
Fixed. Closing.

closed 2021-02-04 by rtyr