prusa3d/PrusaSlicer #3093: Over-Extrusion with 3D Honeycomb infill and small layer height

piotrekm7, 2019-10-22:
### Version
2,1.0

### Operating system type + version
Windows 1-

### 3D printer brand / version + firmware version (if known)
Hypercube, Marlin 2.1

### Behavior
When printing with 0.05 mm layer height and 3D Honeycomb infill type there is huge over-extrusion.

Steps needed to reproduce the problem:
Print anything with 0.05 layer height and 30% 3D Honeycomb infill.

_Is this a new feature request?_

[image]

piotrekm7, 2019-10-22:
It works fine if I use Honeycomb without 3D.

Jebtrix, 2019-10-22:
I can also reproduce this print result including with latest git master.

[image]

supermerill, 2019-10-31:
also duplicate of #1727

rtyr, 2021-02-04:
Duplicate of #1727. Closing.

closed 2021-02-04 by rtyr