prusa3d/PrusaSlicer #3094: support_material_threshold (aka Support Angle) is ignored

GWdd, 2019-10-22:
### Version
2.1.0

### Operating system type + version
Win 7 Pro x64

### 3D printer brand / version + firmware version (if known)
Prusa MK3

### Behavior
Prusa Slicer Support Angle is often ignores, misused, otherwise useless

When selecting 45 degrees, the expectation is surfaces at angles of 45 degrees or more (90 being perpendicular) shall be supported.  

Using the default 55 degrees, I prepared a test model with angles every 10 degrees.  80 down to 10 (10 almost parallel with the bed).

I applied default 55 and then 35 degrees and sliced the model. Both slices created the same support structure; neither anywhere close to the selected angles.

[image]
[image]

* _Steps needed to reproduce the problem_

Here is a package showing the model and setup.
[Angles.zip](https://github.com/prusa3d/PrusaSlicer/files/3758123/Angles.zip)

* _Expected Results_
Setting 55 should support angles of 55 degrees and higher.

* _Actual Results_
Currently, supported angles seem random.

 * _Screenshots from __*PrusaSlicer*__ preview are preferred_
Attached

_Is this a new feature request?_
No.

#### Project File (.3MF) where problem occurs
Attached above.

Jebtrix, 2019-10-23:
If your testing the entire test piece why do you have **Support on build plate only** selected?

Jebtrix, 2019-10-23:
Not saying Support generation doesn't need refinement but I'm not quite following you.

90
[image]
45
[image]
25
[image]
10
[image]

GWdd, 2019-10-23:
If I left build plate only on, my bad. Sorry.
But in the examples above, why does the 80 degree leaf have support when the angle is set to 60?
Same can be asked about the 80, 70, when set to 60?  And why doesn't the 60 have support in this case?  Seems P.slicer is skipping the desired angle.

Yes, my example was flawed by a typo: but the problem is there and reproducible.

Jebtrix, 2019-10-23:
> If I left build plate only on, my bad. Sorry.

What threw me off wasn't so much your project setting alone but your screenshots _also_ displayed from build plate setting. I wasn't sure this if this was a tunnel vision induced chair to keyboard interface error otherwise I'd not have commented :)

> But in the examples above, why does the 80 degree leaf have support when the angle is set to 60?

That support artifacting is the "refinement" I was referring too. After dealing with it so much I don't find artifacts as random anymore, but I understand. There are a bunch of reported issues where this surfaces in different ways.

After double checking the angles in the mesh itself and reviewing results more carefully you are correct. There is multiple issues being exposed here. Some mesh angles require excessive value over compensation in the slicer to produce full support generation (apart from the artifacts). That 60 degree value to get your 40degree feature supported is way excessive. It's not even fully supported till 80 degree value. I'm accustomed to workarounds and nudging things but 20-40 degrees...  I digress. I really like your model, that's a nice test file for the devs.

blibenson, 2019-10-28:
Very similar results from my testing.....basically the support overhang threshold is pretty much useless it appears:

I built a set of blocks in Fusion 360 that go from 0 degrees up to 60 degrees in 5 degree increments measuring the angle from the build plate. I have also run the identical test using overhang tests in Thingiverse. I need to set the "Overhang threshold" to the following values to trigger the respective block:

21 degrees = 15 degree block
26 degrees = 20 degree block
31 degrees = 25 degree block
36 degrees = 30 degree block
44 degrees = 35 degree block
51 degrees = 40 degree block
57 degrees = 45 degree block
62 degrees = 50 degree block
66 degrees = 55 degree block
71 degrees = 60 degree block

trawlergeek, 2020-05-27:
And this problem continues on and on. When is it going to be fixed???

I am abandoning PRUSA SW from this day forward until these OLD and very ANNOYING problems get fixed. It is embarrassing to say I use the Prusa slicer. What a joke. Pathetic, in fact.

lukasmatena, 2020-05-27:
Well, the decision between being told off for not fixing old issues and for not implementing new features is not easy.

blibenson, 2020-05-27:
lukasmatena, while I completely agree, this has been broken for a long time and is a fundamental aspect of any decent slicer.

trawlergeek, 2020-05-27:
Get used to it.  According to online discussions, Your rep is getting worse every day.  For fundamental issues, you decide to fail?  New features when the basics are lacking? Seriously? 

Good luck.  I hope you are free of Covid.

trawlergeek, 2020-05-27:
One more comment:
1393 issues remain open. Has anyone "aged" the issues list to see what is being ignored?
The software manager should be seeing the issues using multiple dashboards. The fact that you probably have issues that are YEARS OLD reflects the lack of professional knowledge of the software manager or equivalent. The backlog needs to be actively managed. Right now, it is a holy mess. 

Your product backlog needs someone (or a team) to review and classify the issues according to "value to the user, when resolved." Introducing new features CANNOT preempt resolution fundamental tool issues. That is just wrong thinking.

What do you think would happen if this tool was "fee based?"  You would stop offering the tool since it would generate NO FUNDS.

You need to carry the message up the "chain" and if they decide that fixing fundamental issues is not in their best interest or is mundane and not as satisfying as creating new features, then we have our answer and a lot of people can stop wasting their own personal time struggling with the sub-standard behavior of a slicer that is free.

If you think you are being "told off," you need to send a message to your supervisors. You aren't being "told off." But the supervisor or director responsible for this "free product" is being informed: they are not doing a good job of managing the product. And it affected the perceptions people have of Prusa.

Enough said...  Moving on.

GWdd, 2020-05-27:
Having issues like this - especially when they are very simple MATH errors a grade school student can understand, implies what the caliber of the coders are.

If they are making these types of mistakes, what else are they doing that makes the tool less useful?