prusa3d/PrusaSlicer #3097: Holes in complex vases

EvilGremlin, 2019-10-24:
### Version
2.1.0

### Operating system type + version
irrelevant

### 3D printer brand / version + firmware version (if known)
irrelevant

### Behavior
Bug in vase post-processor. Holes, whole layers missing.

#### Project File (.3MF) where problem occurs
[vasedrawers_v1.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3765918/vasedrawers_v1.3mf.zip)

Jebtrix, 2019-10-24:
This slices fine with git master so fix is in for next release.

bubnikv, 2020-03-15:
please test with PrusaSlicer 2.2.0-rc3. We have improved the vase mode generator.

rtyr, 2020-05-18:
Seems to be OK in 2.2.0. Closing.

[image]

closed 2020-05-18 by rtyr