prusa3d/PrusaSlicer #3098: First Layer seems to use layer height as nozzle size?

SafSys, 2019-10-24:
### Version
2.1.0+win64

### Operating system type + version
Windows 10 Enterprise

### Behavior
When setting a larger nozzle size the first layer is a lot 'finer' than the upper layers, and has closer together fill lines and outer perimeters. This means that when printing the first layer elephant foot is more pronounced based on the difference between the layers. This can be very pronounced if using a large nozzle - think 1mm+.

This is an image of a 0.3mm 1st layer
[image]

Same sliced object, but the 2nd layer - you can see there's a difference in width
[image]

Exaggerated using a 3mm nozzle - you can see the first 2 layers together.
[image]

Exaggerated by using a 0.05mm first layer.. you almost can't see the perimeters or the lines.
[image]

Jebtrix, 2019-10-24:
It would be helpful if you attach a zipped 3MF project file to review settings in play here. I'm guessing your using auto extrusion widths?

SafSys, 2019-10-25:
[World Scout Badge 2 Negative Block(1).zip](https://github.com/prusa3d/PrusaSlicer/files/3770840/World.Scout.Badge.2.Negative.Block.1.zip)

Filament Extrusion multiplier is 1, Default extrusion width is set to 0 (auto)

closed 2019-10-25 by SafSys