prusa3d/PrusaSlicer #3099: Building fails on make -j<N>

nordurljosahvida, 2019-10-24:
### Version
HEAD

### Operating system type + version
Ubuntu 18

### 3D printer brand / version + firmware version (if known)

### Behavior
Building fails on `make -j<N>` [in my case `-j8`] with:

```
[...]
(.text._ZN5boost6locale4conv4impl14uconv_from_utfIcE7convertB5cxx11EPKcS6_[_ZN5boost6locale4conv4impl14uconv_from_utfIcE7convertB5cxx11EPKcS6_]+0x333): undefined reference to `ucnv_close_60'
(.text._ZN5boost6locale4conv4impl14uconv_from_utfIcE7convertB5cxx11EPKcS6_[_ZN5boost6locale4conv4impl14uconv_from_utfIcE7convertB5cxx11EPKcS6_]+0x348): undefined reference to `ucnv_close_60'
(.text._ZN5boost6locale4conv4impl14uconv_from_utfIcE7convertB5cxx11EPKcS6_[_ZN5boost6locale4conv4impl14uconv_from_utfIcE7convertB5cxx11EPKcS6_]+0x355): undefined reference to `ucnv_close_60'
(.text._ZN5boost6locale4conv4impl14uconv_from_utfIcE7convertB5cxx11EPKcS6_[_ZN5boost6locale4conv4impl14uconv_from_utfIcE7convertB5cxx11EPKcS6_]+0x415): undefined reference to `ucnv_close_60'
(.text._ZN5boost6locale4conv4impl14uconv_from_utfIcE7convertB5cxx11EPKcS6_[_ZN5boost6locale4conv4impl14uconv_from_utfIcE7convertB5cxx11EPKcS6_]+0x488): undefined reference to `icu_60::UnicodeString::~UnicodeString()'
(.text._ZN5boost6locale4conv4impl14uconv_from_utfIcE7convertB5cxx11EPKcS6_[_ZN5boost6locale4conv4impl14uconv_from_utfIcE7convertB5cxx11EPKcS6_]+0x4b0): undefined reference to `ucnv_close_60'
(.text._ZN5boost6locale4conv4impl14uconv_from_utfIcE7convertB5cxx11EPKcS6_[_ZN5boost6locale4conv4impl14uconv_from_utfIcE7convertB5cxx11EPKcS6_]+0x4c5): undefined reference to `icu_60::UnicodeString::~UnicodeString()'
(.text._ZN5boost6locale4conv4impl14uconv_from_utfIcE7convertB5cxx11EPKcS6_[_ZN5boost6locale4conv4impl14uconv_from_utfIcE7convertB5cxx11EPKcS6_]+0x4d0): undefined reference to `ucnv_close_60'
collect2: error: ld returned 1 exit status
src/CMakeFiles/PrusaSlicer.dir/build.make:127: recipe for target 'src/prusa-slicer' failed
make[2]: *** [src/prusa-slicer] Error 1
CMakeFiles/Makefile2:218: recipe for target 'src/CMakeFiles/PrusaSlicer.dir/all' failed
make[1]: *** [src/CMakeFiles/PrusaSlicer.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
```

I don't understand why building is so extremely under-documented. I spent about 3H installing all required dependencies suggested from slic3r wiki, then had to build TBB from source, figure out where to place the cereal lib, then the widgets, and nowhere does it even remotely mention to do this. Am I missing something fundamental here? Can I contribute to some better documentation?

But most importantly, how do I fix this?

[Feature request corner]
1. Nightly builds. The error I'm looking to fix has been implemented a month ago just after 2.1 stable release. A nightly build would completely solve my problem in a snap for a print I need done by tomorrow.

Thanks so much for the great work.

vojtechkral, 2019-10-24:
@nordurljosahvida Building on Linux is somewhat documented at [doc/How to build - Linux et al.md](../blob/eee20e23e199989dfde0f963bacd936805f6a2ed/doc/How%20to%20build%20-%20Linux%20et%20al.md), but of course, there's definitely room for improvement.

As for your issue, it seems boost can't find libicu, but you might have a look at the doc first (unless you already saw it)...

nordurljosahvida, 2019-10-24:
@vojtechkral thank you so much, this of course is intended as very constructive criticism, and i love the project. that was my base for building, but i found that - i think - all the extra steps are not mentioned there. unless i'm understanding something wrong in the article and all deps are installed automatically.

lukasmatena, 2019-10-25:
@nordurljosahvida What are the "extra steps" that you are referring to? Both me and @vojtechkral have incidentally built PrusaSlicer on a fresh Debian installation in the past week. As far as I remember, it is really only about cloning the repo, building the dependencies by the prepared script and then linking to them statically when building PrusaSlicer. There were no external dependencies apart from GTK and OpenGL related ones (libgtk2.0-dev libgl1-mesa-dev libglu1-mesa-dev), which is probably the only thing missing in the documentation. It can get painful if you need to build Perl unit tests as well, but I assume that's not what you were doing.

You seem to have run into many more issues. Could you please be more explicit and tell us what they were? Pasting a cmake output could also help.

The error message you posted seems to be a complain about absenting ICU library, which is boost's dependency. According to https://www.boost.org/doc/libs/1_66_0/libs/regex/doc/html/boost_regex/install.html , boost should be able to recognize its absence and compile without it (which it probably did), but PrusaSlicer apparently relies on some of its functionality (I myself don't know where). If that's so, installing the library from your package manager and rebuilding boost would solve it. I guess the package name for Ubuntu 18.04 would be `libicu60`,  but you better judge it yourself.

I just checked that the ICU library was already bundled on the Debian I built PrusaSlicer on last week. Maybe we should add the library into the dependency list.

tamasmeszaros, 2019-10-29:
Could be a cause of problems when you mix statically built dependencies and distro provided dev packages. Also, very confusing that it does it with `-j<N>`. We normally build with [ninja](https://ninja-build.org/). You could take a chance with it. The debian package is called `ninja-build` .

nordurljosahvida, 2019-10-30:
@lukasmatena @tamasmeszaros thank you very much for the feedback. Extra steps [installing all required dependencies suggested from slic3r wiki, then had to build TBB from source, figure out where to place the cereal lib, then the widgets] were most likely due I think to the fact that - not being technical in building things - I have no idea what `building the dependencies by the prepared script and then linking to them statically when building PrusaSlicer` means. If that is what's mentioned in the linux howto, then I'm missing something, as I sort-of imagined, and some example commands could really work. Following the steps exactly [create the build dir, cd into it, etc...] did not in any way build those dependencies. Can I ask you for the steps you follow when building from debian? I've installed ninja-build, can I ask you for the commands using ninja as well? Thanks again

lukasmatena, 2019-10-31:
@nordurljosahvida 
Just a note on @vojtechkral 's answer. In order to use ninja, both cmake calls might need an extra parameter `-G"Ninja"`. CMake produces `make` makefiles by default (or at least on my system).

However, I'm myself always building with `make` and have no issues.

nordurljosahvida, 2019-11-01:
Amazing thanks, this worked! Can we implement this into the docs?

vojtechkral, 2019-11-04:
@nordurljosahvida Perhaps there could be a TL;DR section on the top of the How tos, but I'm sort of concerned people will just try out the commands without actually knowing what they are doing and then report issues when something goes wrong without reading the rest...

closed 2020-03-16 by tamasmeszaros

nordurljosahvida, 2020-03-17:
@vojtechkral Ok, I understand, however i do think there should be some building instructions, even at the bottom of the how tos, otherwise non-technical users won't be able to build this and test things out.

There are long periods between alpha releases and 600 commits could really make the difference between getting a print done and PS crashing indefinitely.

Thanks

takanotume24, 2020-03-25:
Hello.
Thanks for sharing the build method.
When I tried it on Ubuntu18.04, I needed the following additional packages
```
$ sudo apt install libgtk2.0-dev libudev-dev ninja-build
```
I'm unfamiliar with CMake etc. so I'm not sure, can this be added to a dependency such as /deps/deps-linux.cmake? (except ninja-build).

nordurljosahvida, 2021-08-23:
Hi, I've built again after some time and cannot understand how to compile for GTK3. Ideal would be also GTK3 appimage, but even by running `./prusa-slicer` after building would be enough. Any pointer on how to do this?

bubnikv, 2021-08-24:
Add the following parameter to cmake invocation:
-DSLIC3R_GTK=3

nordurljosahvida, 2021-08-24:
Very good, thanks. I imagine building the appimage instead is a completely separate process?

So do I append the value as such?

cmake .. -G Ninja -DSLIC3R_STATIC=1 -DSLIC3R_GTK=3 -DCMAKE_PREFIX_PATH="$PWD/../deps/build/destdir/usr/local"

Because that throws this error:

```
CMake Error at src/CMakeLists.txt:47 (message):
  

  Could not find wxWidgets 3.1.

  Hint: On Linux you can set -DSLIC3R_WX_STABLE=1 to use wxWidgets 3.0
```

lukasmatena, 2021-09-02:
@nordurljosahvida The invocation is correct. What is probably the problem here is that you most likely build wxWidgets against GTK2, so  cmake does not consider them worthy. Rebuild dependencies with `-DDEP_WX_GTK3=ON`. I have added this information into https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Linux%20et%20al.md.

nordurljosahvida, 2021-09-02:
That worked, thanks! I'm continuing the conversation here: https://github.com/prusa3d/PrusaSlicer/issues/4959 to see if we can get everything into a neat command set. For me until a few weeks ago "add this option when building dependencies" and seeing two commands with cmake and other two with ninja made absolutely no sense and I would have had to test at least 8 different variants of start-to-finish commands. Having a simple, start to finish guide, with a single "best-practice" [if this exists and makes sense, I'm wondering] way, or alternatively a single way that is known to work, is essential for us non-devs, if this is a target audience for alpha/beta testers you'd like to reach of course. Thanks for all the help in this ticket. Closing as solved.

lukasmatena, 2021-09-02:
> For me this issue is solved.

Glad to hear that. Let's continue the discussion in #4959, like you said.