prusa3d/PrusaSlicer #3101: Feature request: move nozzle diameter from printer to print settings

langph, 2019-10-24:
### Version
2.1.0
### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
MK3S MMU2S

### Behavior
When I change the nozzle diameter to 0.6 , I have to use a different printer. That is not logical and irritating if you are using octoprint - you have to register it for each printer. It is more logical to put it under print settings.

_Is this a new feature request?_ yes

#### Project File (.3MF) where problem occurs

Jebtrix, 2019-10-24:
A more likely solution would be global Print Host Upload setting like issue #3050

langph, 2019-10-25:
> A more likely solution would be global Print Host Upload setting like issue #3050

Why more likely? From a users perspective? The nozzle size has the most to do with parameters in printer settings, for example: extrusion width.

Jebtrix, 2019-10-25:
I was referring to re-entering print host settings. More likely because moving the nozzle setting would be a lot more changes code wise. Then dealing with existing profiles and older saved projects users have. I just don't see the devs doing it to solve that particular problem that's all.

pepelevamp, 2019-11-25:
well its changing print hardware. i think its valid to have it under printer settings.

bobstro, 2020-02-24:
For the love of god, DO NOT make this a change. At best, an override. I count on being able to use various print settings with a variety of nozzle sizes. The unlinking of printer, print and filament settings is one of the best features of PrusaSlicer and sets it apart from others. I don't want to have to create a dozen print settings variants for every possible nozzle size I use.

langph, 2020-02-24:
> For the love of god, DO NOT make this a change. At best, an override. I count on being able to use various print settings with a variety of nozzle sizes. The unlinking of printer, print and filament [...]

But there already are print settings for each nozzle size! Choose a 'printer' with 0.6 nozzle and see different print settings for this 'printer'.

bobstro, 2020-02-24:
> But there already are print settings for each nozzle size! Choose a 'printer' with 0.6 nozzle and see different print settings for this 'printer'. What could be useful, is a filter on nozzle size, so you see only the print settings with the nozzle you installed on the printer.

And this suggestion would require me to create print profiles for each and every one of those. As it is now, I have a collection of about 20 print profiles for different layer heights and quality. By setting the extrusion widths to 0 (auto) I can use each of these with any of my nozzle profiles. I've got roughly 24 nozzle profiles, corresponding to different sizes and print qualities.

If this suggestion is implemented, I'll have to worry about creating 480 different profiles simply because you want to hard-code the nozzle size into print settings.

The ability to separate print, printer and filament settings is a strength with PrusaSlicer. This suggestion breaks that to work around a limitation in Octoprint that not everybody cares about. I use Octoprint with different nozzle and print profiles all the time without making a change.

pepelevamp, 2020-02-25:
dear god can you please not move nozzle slice into print settings.

if prusaslicer wants to kiss the ass of octoprint then they might as well
ship a raspi with the prusa printers.
they already revoked usb printing / built in controller stating that
'octoprint did it better'. the truth is that prusa slicer was caving to
silly windows COM port problems that if you didnt use windows, weren't a
problem.

this kind of stuff is what raises big red flags for prusa for me. decisions
that screw with people's workflows plus the fact that they want to build
empires and platforms to compete with what already exists.
prusa should just make good printers and leave the slicer stuff to other
people.

bubnikv, 2020-02-25:
>  and leave the slicer stuff to other people.

to whom?

bobstro, 2020-02-26:
> and leave the slicer stuff to other people.
> to whom? 

Nobody! Prusa is doing great work with PrusaSlicer. Please keep the focus on maximum flexibility for the people slicing for printers.

bubnikv, 2020-02-29:
Thanks for suggestion, but we will not do that. 
Closing.

closed 2020-02-29 by bubnikv

pepelevamp, 2020-03-01:
if i was going to be more specific with my answer i would have been. other
people.
i'm tired of the dickhead-moves from prusa in on the software side of
things.

the printers are good. the software is evil. too many red flags indicative
of a tyrant.

pepelevamp, 2020-03-01:
like USB printing? yeah flexibility alright.
and having your location 'randomized' and put on a world map for everyone
to see? yeah, they really have your best intentions at heart.
the printers are good. i have one. but they have proven many times over now
that they are more than happy to just screw up what you depend on without
any care for your needs, & just give you bullshit answers.

langph, 2020-03-01:
> Thanks for suggestion, but we will not do that.
> Closing.

How rude ;-). Can you please tell me why? From a functional perspective: if you change the nozzle on your printer, it is not suddenly a new printer. I do not care if you follow my exact suggestion, but there has to be a way to make this more user friendly and more understandable.

langph, 2020-03-01:
> like USB printing? yeah flexibility alright. and having your location 'randomized' and put on a world map for everyone to see? yeah, they really have your best intentions at heart. the printers are [...]

Now that is a power-rant :-). If I am correct, Prusa software is open source. You can fork it and change it yourself and/or find like-minded persons to form a new user group with developers.

pepelevamp, 2020-03-03:
Thanks :) I have lots of rants.

Re forking: I know I can fork, but that then becomes a huge burden. I would
be constantly working against yet within the whims of their codebase. Plus
it's all the outside stuff like marketing & commerce that often dictates
use numbers in the free software world and standards & vendor support
reigns king.

Plus prusa are creating platforms & standards everywhere & getting people
to adopt stupid norms. It's difficult to take on something like that.

Plus I don't wanna use prusaslicer it's shit. Slic3r was archaic but to the
point and didn't want to become an ~ experience ~

bubnikv, 2020-03-04:
@langph 
> pepelevamp Plus I don't wanna use prusaslicer it's shit. Slic3r was archaic but to the
point and didn't want to become an ~ experience ~

Exactly. Upstream slic3r is the right product for you. You can exercise your communication skills there, they will certainly appreciate it.

pepelevamp, 2020-03-06:
was referring to slic3r prusa edition.
but you make a valid point of investigating stock slic3r. while its still
around