prusa3d/PrusaSlicer #3103: Slicer 2.1 ignoring parts of product

premek81, 2019-10-25:
Version 2.1.0
Windows 10 & MacOS X
Original Prusa i3 mk3### Behavior

When importing previously used STL, it first looks OK, but after slicing, part of the model is missing. Wall thickness is the same as section underneath it (1mm minimum wall, 2mm ceiling, both are missing, the wall continues to the right, also missing above the ceiling part), when sliced in 2.0.0 (same config bundle) or Simplify3D, the same model slices correctly. 
First picture is from "3D Model" view, second is Preview, after slicing.

[image]
[image]

FidelCapo, 2019-10-25:
Can you please attach 3mf or STL file? Thank you

Jebtrix, 2019-10-25:
This is another model that slices fine if you just merge dup vertices.

[Lepicky - spodek_merged_vertices.zip](https://github.com/prusa3d/PrusaSlicer/files/3772754/Lepicky.-.spodek_merged_vertices.zip)

Salamandar, 2019-10-25:
I have the same problem on multiple parts. Small features are lost, even though they are not that small.
[image]
[image]

To get those lines printed I had to set a +0.1 XY size compensation… Or they are not printed AT ALL.

And also : 
[image]
[image]

premek81, 2019-10-25:
Thanks, I had a feeling that it should be possible to modify the model to slice correctly, however my issue was that previous versions of PrusaSlicer sliced it correctly without any modifications.

Jebtrix, 2019-10-25:
Actually I'm poking at this a little more. This is more than merging duplicate vertices. The exported model I did was from Blender 2.8 in binary STL format. Exporting the same model as ASCII I'm getting **_nothing_** showing up in Preview on latest git master. 

[Lepicky - spodek_merged_vertices_ASCII.zip](https://github.com/prusa3d/PrusaSlicer/files/3772816/Lepicky.-.spodek_merged_vertices_ASCII.zip)

Jebtrix, 2019-10-25:
@Salamandar Both your examples are more to do with extrusion widths not fitting the geometry. Especially the first example with the narrower width on that angle turn of the feature. Lowering the extrusion width either globally or strategically with a modifier would get a slice. XY compensation is basically going the other direction to do the same thing though I'd favor adjusting extrusion width before I'd do that. I could select 0.05 layer height and use 0.051 on every extrusion width setting and that makes no difference with @premek81's issue.

Jebtrix, 2019-10-25:
One last tidbit. Using meshlab and just merging close vertices at 0.0001 unit scale threshold, a single vertice is merged. Exporting both ASCII and binary STLs from meshlab both slice fine then.

[Lepicky - spodek_meshlab_1vertice-merged_binary.zip](https://github.com/prusa3d/PrusaSlicer/files/3773611/Lepicky.-.spodek_meshlab_1vertice-merged_binary.zip)
[Lepicky - spodek_meshlab_1vertice-merged_ascii.zip](https://github.com/prusa3d/PrusaSlicer/files/3773612/Lepicky.-.spodek_meshlab_1vertice-merged_ascii.zip)

premek81, 2019-10-27:
I have a new files that are causing problems for 2.1.0 but not for 2.0.0... This time it is with extremely simple models, again from OpenSCAD (This time I attached the original SCAD files). 
"No layers are detected"... And at first with flag 2 it was "extremely thin extrusions detected".

[Flag.zip](https://github.com/prusa3d/PrusaSlicer/files/3776247/Flag.zip)

Again, probably bad model from openscad, but 2.0.0 slices correctly and the resulting gcode prints as expected... (It is the difference at the //end comment that is causing it)

Jebtrix, 2019-10-28:
Same solution for both of those. Merging vertices at distance threshold of 0.0001 unit size. I have to change my nomenclature since older Blender called that function simply remove duplicate vertices. Now its called Merge by Distance. These models have actual duplicate vertices (reported by Meshlab) but they do not slice correctly unless merging by distance is performed which usually ends up merging an additional 1-6 vertices. 
So to @bubnikv all those models solved by merging vertices I marked in github, take note its the merge by distance 0.0001 as the _real_ solution.

FidelCapo, 2021-02-01:
This issue should be fixed in next release.
Closing.

closed 2021-02-01 by FidelCapo

Salamandar, 2021-02-01:
@FidelCapo is there a pull request you can link this issue to ?

FidelCapo, 2021-02-01:
@Salamandar Do you mean commit? https://github.com/prusa3d/PrusaSlicer/commit/9c5dd358ab23e9351dce0b9c71eaa5ddf92d6f17

Salamandar, 2021-02-01:
> 
> 
> @Salamandar Do you mean commit? [9c5dd35](https://github.com/prusa3d/PrusaSlicer/commit/9c5dd358ab23e9351dce0b9c71eaa5ddf92d6f17)

Thanks !

FidelCapo, 2021-02-01:
@Salamandar But as @Jebtrix said. Your issue on screenshots is probably different issue than original one. I test if issue from first post is fixed and it is.