prusa3d/PrusaSlicer #3104: Concentric/archimedean Infill gaps and odd top/bottom concentric pattern inside concentric infill

Boorkus, 2019-10-26:
### Version
PrusaSlicer-2.1.0+win64-201909160915

### Operating system type + version
Win10 64b

### 3D printer brand
Prusa i3 mk3 **0.6mm nozzle profile**

### Behavior
Having two odd issues with concentric infill and with archimedian infill leaving large gaps/odd patterns for a particular STL generated from a solidworks part. Behaviour occurs on both 0.6mm and 0.4mm nozzle profiles, but the 0.6mm profile makes it more obvious due to the larger extrusion widths. 

1:
Concentric bottom infill has a gap, (exactly like this issue https://github.com/prusa3d/PrusaSlicer/issues/577) where it just doesn't fill for several concentric loops, see attached image:
[image]

2:
Concentric top infill does weird solid infill(?) radial loop patterns halfway through the flat surface and ruins the otherwise smooth surface finish. Also behaves oddly on the outer rim. I noticed the stl elements make the object look "like a pie" so I think that has something to do with the "pie slices" confusing the concentric fill algorithm? Can't tell if those are gaps from the previous solid infill layer below showing through, or if it's actually adding that solid infill and ruining the surface finish.
See attached:
[image]

3:
Trying to use archimedean infill pattern to fix the problem *mostly* works, except it leaves big gaps near the perimeters on the outer rim, and leaves a nasty half-spiral-gap on the small diameter hole (10mm) in the centre where the solid infill peeks through the gap. The printed part looks very gappy and not acceptable for the job I needed it for. 
See attached:
[image]

I have tried to include .3MF files and STL for both concentric and archimedean examples but github is bugging out and won't process the upload, somethign to do with the winrar/github HTML bug... 
apologies for the stupid google drive link but it's literally all that I can get to work. I can email the files through if someone contacts me with an email address to use...

https://drive.google.com/file/d/10KCXxdM7Qnidov67JZXoy2yPaD67bDG6/view?usp=sharing

tomrozb, 2020-03-21:
@bubnikv any chance to get this fixed? It was reported several times since 2018 and still the concentric pattern is almost unusable for top/bottom fills. It doesn't look good on prints as it leaves huge gaps in random places.

[image]

bubnikv, 2020-03-21:
Sure, we will fix everything one day.

It would be a greatest help to the developers, if the load of
unnecessary new issues and basically support requests flooding PrusaSlicer
github ceased, so we developers can concentrate on development and bug
fixing.

GarthSnyder, 2020-04-30:
Here's another, reproducible example of gaps in concentric top fill. The gap does appear in the actual print. See layer 64.

[ConcentricGap.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/4561197/ConcentricGap.3mf.zip)
[image]

haarp, 2021-12-14:
Similar issue with Archimedean Chords. Once the remaining space falls below the extrusion width, the slicer will simply stop filling the space, leaving a visible gap.

[image]

[knob_knurled.zip](https://github.com/prusa3d/PrusaSlicer/files/7710837/knob_knurled.zip) (rename to .3mf)

The chords could and should be continued longer, even if the space appears to not be enough. Probably until the gap falls to 50% of the extrusion width. It would still look much better than not filling the gap at all.

To improve this further, it could be considered to reduce the extrusion width dynamically and continuously to finish the spiral.

Thanks for your work on PS!

dartrax, 2021-12-20:
Duplicate of #577 and (point 2 an3 of first post) sounds like #4412 etc.

men8ifr, 2022-02-06:
I have this issue too...

bubnikv, 2022-06-29:
Implemented in PrusaSlicer 2.5.0-alpha2. 

In PrusaSlicer 2.5.0-alpha2 we have ported the Arachne perimeter generator from Cura. Thanks Ultimaker and Cura team for keeping Cura open and for their dedication.
With Arachne enabled for perimeters, the Arachne is newly used to fill in concentric infill completely.

Closing.

closed 2022-06-29 by bubnikv

haarp, 2022-07-20:
@bubnikv `Perimeter generator` is under `Layers and perimeters`. Could you add a note in the PS that generator also affects concentric infill? It makes sense, but isn't immediately obvious.

nmattia, 2024-03-11:
@Boorkus did the fix resolve your issue for Archimedean Chords infill as well? I'm having issues on PrusaSlicer 2.7.2 with Arachne enabled. I still have a spiral-y gap near the perimeter:

[image]

Boorkus, 2024-04-08:
@nmattia Just tested this out with PrusaSlicer 2.7.4 with Arachne. 
Concentric infill now behaves properly, but Archimedean Chords still does not. I get exactly the same as your image, it does not fill all the way to the edge in the slicer preview

carlosjln, 2024-04-28:
Dang it... I'm stumbling upon this issue now trying to optimize a part for production and I thought the Arachne option would deal with it.

[image]