prusa3d/PrusaSlicer #3106: Can't slice a specific model that works fine in cura

JimmySnails, 2019-10-27:
### Version
2.1.0+win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Original Prusa MK3S (but the problem is in the slicer)

### Behavior
* Just add the model and slice it. Happens with each profile. 
Bottom Part of the head (the ring) is missing after slicing and an error message pops up. (see screenshot below)

Works fine in cura and chitubox, but doesn't work in Prusa slicer. I'm wondering if this is a bug or a faulty model.

* This is the model:

<https://www.thingiverse.com/thing:2605388>

* Error Message:

[image]

Here is the AMF file:
[Led_Head.zip.amf.zip](https://github.com/prusa3d/PrusaSlicer/files/3775640/Led_Head.zip.amf.zip)

Jebtrix, 2019-10-28:
[image]
[image]

Jebtrix, 2019-10-28:
Cura's thin wall detection works relatively fine. PrusaSlicer's currently does not. Your geometry is smaller than your extrusion widths. Simply reduce the extrusion widths so it actually fits the geometry.  You could turn on Thin Wall detection but the result is messy compared to doing it this way.

JimmySnails, 2019-10-30:
Thank you, but is it a good idea to set the the layer width to be smaller than the actual nozzle size? 
Gonna try printing it that way!
I can't scale the model since it's designed to fit a "Led Earring" in it, so i'm gonna stick with your solution. 
Should i close the issue or leave it open for "Thin wall detection"?

Jebtrix, 2019-10-31:
> Thank you, but is it a good idea to set the the layer width to be smaller than the actual nozzle size?

That is exactly what Thin wall detection does in any slicer. Either it reduces extrude to fit, or doesn't get printed. Its a necessary compromise. When your designing a part yourself you would definitely take that into account to eliminate/minimize thin wall features in the first place (at least for functional parts). Printing downloaded models you just have to make due sometimes.

JimmySnails, 2019-11-10:
Thx for the explanation. Then i'll close this now i know how to adjust things.

closed 2019-11-10 by JimmySnails