prusa3d/PrusaSlicer #3108: Speed setting value resets

gnydick, 2019-10-27:
### Version
Version 2.1.0+win64

### Operating system type + version
Windows 10 (build 18362), 64-bit edition

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
* Go to Print Settings -> Speed -> Perimeters
* Change any of Perimeters, Infill, Support Material, Bridges, or Gap Fill to "0" and when you change focus the value reverts to whatever the last non-zero value was.

#### Project File (.3MF) where problem occurs
It won't upload the zip file :(

glowtape, 2019-10-28:
Can confirm.

Jebtrix, 2019-10-28:
Can reproduce on latest git also.

YuSanka, 2019-10-29:
Thanks for your report. Working on it in [JIRA-1043](https://dev.prusa3d.com/browse/SPE-1043)

gnydick, 2019-10-29:
You're welcome. Just a heads up, there are other fields in other tabs as
well. I just haven't itemized them all.

FidelCapo, 2019-11-04:
This issue should be fixed in next release.
Closing.

closed 2019-11-04 by FidelCapo