prusa3d/PrusaSlicer #3109: Brim is not generated

dbarvik, 2019-10-28:
### Version
2.1.0
Windows 10

### Behavior
I would like to add brim to my object in PrusaSlicer, but it's not working. I have set the brim width to 5 mm and on "main page" of slicer I have ticked the brim, and it's not generated on my object. Every change on that "main page" as print setting, filament, printer, supports, infill brim doesn't take effect in slicing. 
When i wanna change for example layer height, I need to click on object, select Layers and Perimeters and change the value there... My printer profile doesn't take effect in any change.

[image]

Jebtrix, 2019-10-28:
You will need to attach a zipped 3MF project file for people to review. Especially since this is not using default Prusa printer profiles.

dbarvik, 2019-10-28:
Here I attach 3MF project file.
[3DBenchy.zip](https://github.com/prusa3d/PrusaSlicer/files/3780935/3DBenchy.zip)

Jebtrix, 2019-10-28:
I open that project and Print Settings > Brim > Brim width is set to 0. Change it to 5 there and it works fine. This may be the setting resetting issue that is mostly resolved in git master. Try using 5 again and press Enter to make it stick.

dbarvik, 2019-10-28:
That's weird, I have set 5 mm brim there, and when I change value, nothing happened
[image]

dbarvik, 2019-10-28:
I have found out, that if I open PrusaSlicer and import the object, I cannot add brim or change printer profiles. But if I save it in 3mf file, and open it. I can set brim, supports and printer profiles. 

So what can cause the problem, when I open directly PrusaSlicer? It's not comfortable to open it, save project, reopen the project and set profile etc.

JohnnyDeer, 2020-03-16:
Hi, i have same results as @Jebtrix. If i save your 3mf with added brim settings, open new slicer window and import, brim is there.
Could you please try it on slicer version 2.2.0?

MucaroBoricua, 2020-05-13:
I'm seeing the same problem in version 2.2.0.  It's a bug!

closed 2020-05-23 by dbarvik