prusa3d/PrusaSlicer #3110: Layer Slider Artifacting Issues

michaelPotter, 2019-10-29:
### Version
Version 2.1.0

### Operating system type + version
Manjaro Linux
_System info is attached_
[system_info.txt](https://github.com/prusa3d/PrusaSlicer/files/3781403/system_info.txt)

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
* Mousing over the layer slider in the layer preview view causes issues and artifacting all over screen.
[image]
[image]

* **To Reproduce:** Slice a model, enter layer view, mouse over layer slider

_Is this a new feature request?_ No

Jebtrix, 2019-10-30:
Can you provide a little more detail for the devs? What exact desktop environment you using XCFE, Gnome? Wayland? X11?

michaelPotter, 2019-11-07:
FYI I never did get the drivers figured out or wayland working, but I upgraded to the 4.14 kernel for other reasons (was on 4.9) and this issue went away

closed 2019-11-10 by michaelPotter