prusa3d/PrusaSlicer #3116: Purge block printing in air

andrewbecker, 2019-10-31:
### Version
_Version of PrusaSlicer used goes here_
2.1.0
_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
OSX 10.15.1
### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
Prusa MK3s with MMU2s
### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
When there are no tool changes left at the end there are two lines printed in air on the purge block
[image]
[image]

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_
[white_center.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3793331/white_center.3mf.zip)

lukasmatena, 2019-10-31:
This is intentional. See whichever of [FAQ](https://github.com/prusa3d/PrusaSlicer/wiki/Frequently-Asked-Questions-(FAQ)), #2412, #1355, #2088, #1616, #1070, #1268, #1423, #1386, #663, #1581 or #2612.

Please, search the tracker before filing a new one issue. Closing.

closed 2019-10-31 by lukasmatena

andrewbecker, 2019-10-31:
Thanks. Would it make sense not to display this in the code preview?

lukasmatena, 2019-11-01:
It would stop these issue reports from coming, but I don't think it's the correct solution. The ramming extrusions that you see there are also on the wipe tower. It would not be consistent to only hide some of them. Not to mention they are extrusions like any other.

I think that the preview should show what the printer will do, as exactly as possible. Hiding parts of it would also make testing and debugging of the software more difficult.

JohnOCFII, 2019-11-01:
> I think that the preview should show what the printer will do, as exactly as possible. Hiding parts of it would also make testing and debugging of the software more difficult.

As that material DOES get extruded, it should be previewed, I agree.  I wonder if the MMU Handbook should be updated to include the FAQ entry, as it is somewhat disconcerting to see the extruder extruding in mid-air.