prusa3d/PrusaSlicer #3120: Send to Octoprint error "0x80092012" when using SSL

skint007, 2019-11-01:
### Version
2.1.0+win64

### Operating system type + version
Windows 10 Pro 1903 x64

### 3D printer brand / version + firmware version (if known)
Prusa MK3 / FW 3.8.0

### Behavior
I recently started using SSL cert for my octoprint instance and now am getting an error when trying to send a print to octoprint or "test" the connection.
I originally had the IP in there, but have tried putting the DNS to my octoprint in there, along with adding "HTTPS://" in front of both the IP and DNS. Regardless I still get the same error.
It says to make sure that my custom CA is in my cert store, which it is, chrome and my phone both see the url as a secure site.
I'll also add I don't have any basic authentication setup with the haproxy to achieve ssl

### Screenshot:
[image]

nordurljosahvida, 2020-10-03:
+1 here, exact same scenario, client of mine on win10, 2.2.0 stable, we were only able to get the specific error string when running from `prusa-slicer-console-exe --sw-renderer --loglevel=9`, because gui only output a generic error. already tried disabling crl systemwide [not a good idea anyway] and flagging the custom CA's cert as OCSP skippable.

bubnikv, 2020-12-03:
@nordurljosahvida 
What error message did you then get on your console?

> already tried disabling crl systemwide [not a good idea anyway] and flagging the custom CA's cert as OCSP skippable.
Did that help?

I suppose your certificates are simply invalid.

nordurljosahvida, 2020-12-08:
Thanks for looking into this, I don't have the exact terminal output here but the same definition as in the gui shot posted above [the revocation function etc...]. Don't have exact error code here but the definition is identical.

The CA certificate I'm using is working on any other os, it's a fully signed and valid CA that itself signed the cert on the pi, windows recognizes it as valid, chrome as well, just not PS, and microsoft's help on this topic doesn't work systemwide.

bubnikv, 2020-12-08:
Could you please provide with such a certificate?
I suppose you don't want to share yours, but could you generate one, test that it fails and then provide it to us?

nordurljosahvida, 2020-12-08:
This is a cert generated exactly like the non-working one, with a different key of course. It's done by a script so it is functionally identical.
[definitelynottherealone.lan.zip](https://github.com/prusa3d/PrusaSlicer/files/5660671/definitelynottherealone.lan.zip)

rtyr, 2021-02-04:
Is this issue still valid? Does it happen with 2.3.0?

lexitus, 2021-02-04:
Not working for me as of 2.3.0. Using a home-made mkcert CA, imported into the windows cert store.
[image]

SuburbanSpetsnaz, 2021-02-17:
Same problem for me, PrusaSlicer 2.3.0, personal CA for all devices on my network. This cert is recognized by all my machines and my phone as well, but will not work in PrusaSlicer.

[image]

nordurljosahvida, 2021-08-29:
will try asap thanks