prusa3d/PrusaSlicer #3121: Sequential Printing not working #2171

Mazhive, 2019-11-01:
This "problem" has being wandering in my mind quite some time ,....

It could be solved by altering gcode and duplicate the same model gcode after the first but ....

Why cant there be an option that features a pause long enough to take it off and continue with the print so you don't have this issue even with the higher models not having clearance. This would be a great feature i think.

supermerill, 2019-11-04:
What is the difference in printing 2 object in sequential mode wit a pause for manual removal vs launching two prints ?
The second option need two more click, but is it a big deal?

Kachidoki2807, 2019-11-04:
Of course for only 2 objects you can have a doubt on the usefulness.
It is useful when you have a lot of small objects, the sequential printing reduce a lot the stringing between the objects since there are much less travels, and it improve the seam (specially true with the PETG).

If the printer can handle itself the sequential printing of (as an example) 50+ objects of 30min printing time each, it would made my sleeping time much more peaceful. :-)

supermerill, 2019-11-04:
I don't really understand, sorry but i have to play a bit of the "dumb guy".

What is the difference in printing 50 object in sequential mode with 50 pauses for manual removal vs launching 50prints ?
The second option need 50*two more clicks, but is it a big deal?

Please explain a bit more exactly what process you though about vs current process.

Kachidoki2807, 2019-11-04:
The "true" sequential printing means that this is the slicer that arrange the G-Code to print one object at a time (in fact, it will handle the clearance), without any pause between them. This is the major difference with the continuous printing.

By reading again the OP, the request is slightly different, and need a pause indeed. In such case it wouldn't be a big deal to manually restart a print.

Mazhive, 2019-11-05:
this way you can make some kind of automated system easier i think by  
using some kind of a bump system to knock of your print and let it  
fall into a bin or something ..
hope to clearify some . but yes you are right in the sense of a person  
present at the time. it would make no diffrence.
  kind regards peter

bubnikv, 2020-02-07:
Pausing a print to let the user remove objects from the build plate is questionable. If you have an automatic plate exchange system, then your system may swap the G-codes as well.

Closing.

closed 2020-02-07 by bubnikv