prusa3d/PrusaSlicer #3122: Wrong permissions of files/directories in Linux tar.bz2

bablokb, 2019-11-02:
### Version
PrusaSlicer-2.1.0+linux64-201909160925.tar.bz2

### Operating system type + version
Linux

### 3D printer brand / version + firmware version (if known)
N.A.

### Behavior
After unpacking of linux tar.bz2 archive, files and directories have wrong permissions. Normal files should have 644 (i.e. rw-r--r--), directories should have 755 (i.e. rwxr-xr-x). The two prusa-slicer executables should also have 755.

For me as a Linux-guru it is a matter of a few cryptic commands to change this, but this is unnecessary work for every release.

I assume your buildbot just packs the files, so it is probably a matter of changing the permissions of your master codebase once.

#### Project File (.3MF) where problem occurs
N.A.

Bernhard

bubnikv, 2020-03-15:
is it still broken? Clould you please close this issue? Thank you.

bablokb, 2020-03-16:
Yes, this is still broken. Just run

    tar -tvjpf PrusaSlicer-2.2.0-rc4+linux-x64-202003151142.tar.bz2

and you will see that  e.g. the main executable is only executable by root (and you don't run normal programs as root):

    -rwx------ root/root 123889120 2020-03-15 12:50 PrusaSlicer-2.2.0-rc4+linux-64-202003151142/bin/prusa-slicer

Normal users can't even read the files/directories. Permissions are wrong and changing this is probably simple. Just change the permissions and ownership of your master code-base. And don't use root to pack the files.

bubnikv, 2020-03-16:
> Normal users can't even read the files/directories. Permissions are wrong
and changing this is probably simple. Just change the permissions and
ownership of your master code-base.

Thanks for checking. No problem, we can change that.

> And don't use root to pack the files.

It is a docker image. Anyways, what user shall we assign to these files?
AFAIK the user is identified by a mapping an integer through /etc/users.
Now root is always user zero I suppose. How about the others? Are there any
other user names default on all Linux systems with the same user ID
assigned?

bablokb, 2020-03-19:
Except for user root (indeed always zero) there is only one convention: ids below 100 belong to technical system-users, above are normal users.
The user is not the critical part, but the permissions.

Can I find the build-script somewhere? I would provide a patch that changes the permissions before packaging it with tar.

bubnikv, 2020-03-19:
> Can I find the build-script somewhere? I would provide a patch that changes the permissions before packaging it with tar.

No worry, I can replace chmod +x with chmod 777.
Now I need to rebuild that docker image.

bubnikv, 2020-03-20:
@bablokb 

For the non-root, shall I add the write privileges to the non-root?
When tar.bz2 ipping our binary directory tree, where all files have rw privileges (-rw-rw-rw), 
after unpacking the write permission to "all" disapears (-rw-rw-r-).

I am not sure how people will handle the AppImage or the binary blob and I am really confused how the tar handles the privileges. What should we do?

btw we likley have similar issues with our OSX build.

bablokb, 2020-04-02:
I would recommend the following commands before packaging:

    cd path-to-root-of-files
    find . -type f -exec chmod 644 {} \+
    find . -type d -exec chmod 755 {} \+
    chmod 755 bin/prusa-slicer
    chmod 755 ./prusa-slicer

The first find-command will give rw-access to the owner and r-access to all others. The second find command wil give rwx-access to the owner and rx-access to all others (execute is necessary for directories). Depending on how you create the files you only have to do it once or every time before packaging.

Install instructions would be: download tar.bz2 archive and then execute the following commands as root (e.g. prepend sudo)

    tar -xvjpf path-to-archive.tar.bz2 -C /usr/local/lib64
    cd /usr/local/lib64
    [ -l PrusaSlicer ] && rm PrusaSlicer
    ln -s PrusaSlicer-2.2.0+linux-x64-202003211856 PrusaSlicer
    chown -R root:root PrusaSlicer-*
    ln -sf /usr/local/lib64/PrusaSlicer/prusa-slicer /usr/local/bin/prusa-slicer

The third command removes an existing link to the current version of PrusaSlicer if it exists. The fourth command creates the link to the current version. The last command creates a link in /usr/local/bin which is usually in the path, so the executable is found. This command is a one-timer and not necessary when you update to a newer version.

This setup has the advantage that you can switch to an old version of PrusaSlicer just by changing the symbolic link back to the old version (third and fourth command).

bablokb, 2021-01-23:
Issue is still the same with 2.3.0

bablokb, 2021-01-23:
Also the label "awaiting response" is wrong

closed 2026-09-29 by prusaslicer-bot[bot]