prusa3d/PrusaSlicer #3123: Feature: Connect intro-line with skirt, to make small skirts easier to remove

rennne, 2019-11-02:
When I print small parts I have sometimes the issue that it is very hard to remove the skirt.

For that it would be handy, when the intro-line would just continue to extrude into the skirt.

Then it would be possible to remove the skirt with just pull the intro-line away

rtyr, 2020-04-20:
I don't think that this is a realistic request. Intro line is not PrusaSlicer feature, it is printed based on the custom start g-code script. It can be different for each printer, for each user or not being used at all. 
You can increase the number of skirt layers, which will make it easily removable. We are using 3 skirt layers in our print profiles. Easy and more universal solution. What do you think?

[image]

rennne, 2020-04-20:
I found that increasing the skirt height, also increase the chance that the skirt becomes lose an screws up the print, sometimes.
The issue arises especially with thin layers like 0.05mm, where 3 layer only add up to 0.15mm, and its sometimes quite hard to remove it.

But:
I think, if PrusaSlicer would provide the lowest Y point of the skirt this could easily be covered within the "custom start g-code script"

Instead of:

G92 E0.0
G1 X60.0 E9.0 F1000.0 ; intro line
G1 X100.0 E12.5 F1000.0 ; intro line
G92 E0.0

this could be used:
G92 E0.0
G1 X60.0 E9.0 F1000.0 ; intro line
G1 X100.0 E12.5 F1000.0 ; intro line
G1 X[skirt_entry_x] Y[skirt_entry_y] E12.5 F1000.0 ; continue intro line to skirt

When it is not used at all skirt_entry_x = midde_of_buildplate_x and  skirt_entry_y = -1.0 or something kind of that idea

fiveangle, 2020-04-20:
I actually spent some time thinking about this a few years ago and didn't understand why the skirt exists at all rather than just specifying a fixed extrusion volume that the user wants extruded.

The slicer would then calculate an extrusion line orthogonal to the start point of the outermost perimeter, then if it hits a bed edge before the requested volume is extruded, turn to follow the bed edge and keep going until the required volume is reached, then inverse that entire path for the start of the print.  This would leave a single "tag" of filament coming out of bottom perimeter at the start location, which is easily removed.

I'm sure some might take issue with it, but I would much prefer this method since it ensures the start of the print is good.  I don't know how many times the nozzle has lifted from the skirt, only to not start correctly at the actual print start on the perimeter, esp. with PETG and a bed that maybe wasn't absolutely perfectly cleaned.

```
     ___________
     |         |
     |         |________  
     |_________|       | <= Requisite line length to fully prime
                       |    nozzle based on mm^3 vol set in prefs
                       | 
                       ^Print start point along right bed edge
```
I don't want to hijack @rennne 's FR so just upvote this comment if you think it's worthwhile.  If there's enough interest I'll submit a separate FR.

rennne, 2020-04-21:
I think you are targeting a similar fundamental issue. It would rather use no skirt at all, but without it, it sometimes does not deliver good results, in regards of flowrate at the start of printing the object.

Having a line going in the right angle to "the print" or "the skirt" or where "the skirt would be" and have it connected to the "intro line" or having that instead of the "intro line",
sounds to me like very related ideas.

fiveangle, 2020-04-22:
Your issue is [difficulty in removing the skirt], and make no mention of pressure control or otherwise. I fail to see how they are related other than my idea helps with yours if one implements the skirt single-layer then connects it orthogonally to the model (which I am not suggesting since I don’t think a skirt is useful anyway except for use as a draft shield).

Regarding your idea to connect the purge line, you received a reasonable response from @rtyr (which most here would probably agree with). If you are suggesting to change your idea, continue here. Otherwise, maybe close it and play around with your start gcode idea then post back a new issue if you find any success with it ?

rennne, 2020-04-23:
Regarding @rtyr Idea, it actually does not solve the problem:
Increasing skirt height when using 0,05mm layers only 3 adds up to 0,15mm which is so thin that it is still very hard to remove, but every additional skirt increases the chance that the skirt detaches and screws up the print.

My GCode Idea is not executable before "skirt_entry_x" and "skirt_entry_y" or
or "orthogonal_entry_point_x" (any y) or similar is not implemented.

From my point of view: Using the skirt in the first place is mainly about pressure control.
This effect is most prevalent when printing small object with little contact surface to the buildsurface at 0,05mm layer height.

Without a skirt, the pressure in the nozzle is not right and the print tends to detach because the first layer was not good enough. 

> (which I am not suggesting since I don’t think a skirt is useful anyway except for use as a draft shield)

I just would prefer not to use the skirt in the first place, but because of that pressure issue I tend to use skits.

rtyr, 2020-04-23:
To be honest, IMO you are trying to solve an issue which is not an issue at all. Just try more skirt layers as I suggested or disable it completely :-). 

Based on your answer it seems, that you have two opposite issues, too high skirt adhesion and too low skirt adhesion (you are afraid, that it will come loose during the print). Skirt is often used as a draft shield for whole object, so printing +-3 layers really shouldn't be a problem. I guess it might be problem only for users, which are using skirt instead of intro line (not uncommon). I also doubt, that anybody actually uses 0.05mm for first layer like you suggesting. 

In situation, where more layers of a skirt are used (which is needed for textured surfaces), your approach would only complicate things for obvious reasons. Also imagine the scenario, when you put small model to the other side of the bed...

I will leave this open, it is not up to me to decide, but my opinion is that this is not needed and there are more important things to be improved.

rennne, 2020-04-23:
Obliviously I fail to get the problem across

> Based on your answer it seems, that you have two opposite issues, too high skirt adhesion and too low skirt adhesion 

Exactly when dealing with very small prints and 0.05mm layer height this happens both. Sometimes the skirt sticks too well to remove it afterward, the other times is sticks so badly that i gets lose during printing. 

I never experienced any of those issues wich normal layer-heights and and normal sized prints.

There are right now three options, for those special printing conditions:

* No skirt -> sometimes print becomes lose because the pressure is lost during the travelmove form the introline to the start of the first layer
* shallow skirt -> prints ok most of the time. Skirt sticks sometimes so good, that sharp tools are required to remove it, thereby potentially damaging the buildplatte.
* High skirt -> print sometimes fails because skirt gets sometime lose.

So I can choose between risking damaging the build platte or sometimes failing prints

rtyr, 2021-01-26:
We will not implement this. I am closing this issue.

closed 2021-01-26 by rtyr