prusa3d/PrusaSlicer #3124: Crash 2.1.0+win64

R08, 2019-11-02:
### Version
2.1.0+Win64

### Operating system type + version
Win 10x64

### 3D printer brand / version + firmware version (if known)
Marlin 1.1.9 Cartesian
Arduino Mega 2560 ramps 1.4

### Behavior
Crash
[2.1.0win64_crash.zip](https://github.com/prusa3d/PrusaSlicer/files/3800810/2.1.0win64_crash.zip)

1. Select the following profiles from zip.
    - Solidoodle (0.8)
    - PLA (0.8) - Hatchbox Orange
    - Solidoodle (0.8)

2. Load InsetTest.stl

3. Slice

4. PrusaSlicer 2.1.0+win64 will crash :(

#### Project File (.3MF) where problem occurs
Also uploaded .3mf as zip.
[InsetTest.zip.zip](https://github.com/prusa3d/PrusaSlicer/files/3800822/InsetTest.zip.zip)

R08, 2019-11-02:
Has something to do with retract distance set to 0. If I set the retract distance to 0.00001, there is no crash.

lukasmatena, 2019-11-04:
I can reproduce it in 2.1.0, it works fine in current master. It was apparently already solved, though I don't know when.

@YuSanka Could this somehow be related to https://github.com/prusa3d/PrusaSlicer/issues/3108 ?

lukasmatena, 2019-11-04:
According to @YuSanka, this is indeed caused by the same bug as #3108. It was therefore recently fixed in current master, the fix will be in PrusaSlicer2.2.0.

Thank you for reporting it and also thanks to @R08 for helping us find the cause.

Closing.

closed 2019-11-04 by lukasmatena