prusa3d/PrusaSlicer #3125: MMU2S floating wipe tower

markscsmith, 2019-11-02:
### Version
2.1.0+

### Operating system type + version
macOS 10.15.1 (19B88)

### 3D printer brand / version + firmware version (if known)
Prusa MK3S MMU2S with current firmware.

### Behavior
* When slicing an object where the first layer is set to the same material as support, the wipe tower floats off the build plate until non-support layers are reached
* Take an object, place a slab underneath it, float the object over the slab, set the slab to extrude support material, slice with support material.
* Slice fills in wipe tower at all layers.
* Slice shows giant gap in wipe tower, prints the same way

[image]
[image]

#### Project File (.3MF) where problem occurs
[kobold_Skirmisher.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3800862/kobold_Skirmisher.3mf.zip)

lukasmatena, 2019-11-04:
That's interesting. Somehow the slab at the bottom fooled the wipe tower generator. I'll look into it.

However, if the purpose of the slab is to ensure the supports stick to the surface, you might consider enabling raft (Print Settings->Support Material->Raft. That should have more or less the same effect and the wipe tower will print fine. Does it work for you or is there a reason you don't want to do that?

markscsmith, 2019-11-04:
I am indeed using it in place of a raft. What I’ve found is that bed adhesion in BOVH is better when it’s solid and connected like the slab.  Rafts tend to have the lines very far apart and end up peeled off by the nozzle on both the PEI and Smooth beds.  There’s other tinkering I could do, but this seemed easier :)

markscsmith, 2019-11-05:
Additional symptom I forgot about: If I change the material to be the default, or any non-support extruder (1-4) I get the error "Error: WipeTowerIntegration::append_tcr was asked to do a toolchange it didn't expect." and a similar floaty wipe tower.

[image]

[image]

bubnikv, 2020-03-16:
While we were not able to reproduce your issue, it sounds similar to https://github.com/prusa3d/PrusaSlicer/issues/3834

bubnikv, 2020-03-16:
Do you get this issue always? Are you able to reproduce it?
How about with https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.2.0-rc4 ?

lukasmatena, 2020-03-16:
@bubnikv I was in fact able to reproduce this one in 2.1.1, but 2.2.0-rc4 seems fine. Not sure what fixed it. I don't think this will be the same as #3834 though.

rtyr, 2021-02-04:
Fixed. No problem in 2.3.0. Closing.

closed 2021-02-04 by rtyr