prusa3d/PrusaSlicer #3128: Spiral vase mode reverted to regular layers at 58.4 mm

RaceReadyJ, 2019-11-02:
2.1.0 release

Windows 10 current

Prusa MK3, 

### Behavior
Slicing in spiral vase mode, reverted to regular layer height (.2) at 58.4 mm
This model at 102% scaling.  
https://www.thingiverse.com/thing:3418609

Expected spiral all the way up
Again, got regular layers after 58.4.  Also, Preview slider bar compressed area above 58.4 to like last 5 percent of preview slider bar space.

[image]
[image]

Bug, not feature request

[190202_Python_Julia_Bowl_v1_z7_1pt2_0pt5i_30isteps_binary.zip](https://github.com/prusa3d/PrusaSlicer/files/3800937/190202_Python_Julia_Bowl_v1_z7_1pt2_0pt5i_30isteps_binary.zip)

RaceReadyJ, 2019-11-02:
Update.  Closed PrusaSlicer and reloaded same model, no scaling, no color changes, same layer height (.2).  Got same issue, but at 61.2 mm.  Also note, mm height figures are "cut off" somewhat in the slider control area.
[image]

jmd73, 2019-11-04:
Hi!
It seems this model has thin walls. If you turn on the parameter "Detect thin walls" under Print Settings->Layers and perimeters, it will slice better.

I haven't checked if all is correct, but it looks a lot better. Another solution is to change the extrusion width to match the wall thickness.

Windows 3d builder even warns about thin walls on this object.

@bubnikv Would it be possible to get a warning in PrusaSlicer when an object has thin walls, since it seems to be a common problem for people?

bubnikv, 2020-03-15:
Spiral vase has been improved in PrusaSlicer 2.2.0-rc3. Would you please test with that build?

rtyr, 2020-03-28:
Tested with PrusaSlicer 2.2.0 final, it seems OK. Closing.

closed 2020-03-28 by rtyr