prusa3d/PrusaSlicer #3131: 2 extruder single nozzle unable to understand parameter of change tool

marcottt, 2019-11-03:
### Version
2.1.0+win64

### Operating system type + version
windows 10 64 bit

### 3D printer brand / version + firmware version (if known)
Anycubic kossel linear plus, modded 2 material with Y bowden. 
Firmware 1.1.9

### Behavior
Hi all, trying to move to prusaslicer,  I made some test and it work but some parameter and behavior i do not understand (note i'm note english) so looking for help for tuning change material process. 

Explain:
during print, when one color (we say color#1) end the layer and need to change, printer made this movement:

a) go to prime tower (ok)

b) prime some material still #1 (what is this time/mm? need to reduce)

c) retract material #1 some mm (i think about 10) 

d) move empty nozzle on prime tower (why? need to reduce to 0 this time)

e) retract material to park position (ok)

f) move slowly material #2 a bit, i think 20mm (what is this time ? i need to change)

g) move with more speed #2 (what is this time ? i need to change)

h) prime material #2 on prime tower (ok)

Can some one tell e about above movement which parameter are involved ?

Also, what is in "purging volume" the "volume unload" parameter ? changed from 70 t0 10 but i do not see change in print...

I see also a problem on layer 1, when color#1 is finish printer change color and print material#2 but di not elevate nozzle so nozzle bump on already printed part with color#1; this appear only for layer1 (it si normal? i tried "avoid crossing perimeter but no result).

Thanks to all, i hope to not be excessive 

Marco

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs

[5h_pla_6coni_bicolore.zip](https://github.com/prusa3d/PrusaSlicer/files/3801799/5h_pla_6coni_bicolore.zip)

lukasmatena, 2019-11-04:
Most details of the toolchange sequence on the wipe tower can be configured in Filament Settings->Advanced->Toolchange parameters with single extruder MM printers. Read tooltips of each of the settings.

The sequence consists of
- ramming (rapidly extrudes some of the old filament to make sure there is no string after you unload
- unload of the old filament
- cooling of the filament by moving it back and forth
- parking of the old filament
- loading of the new filament
- priming of the new filament (this is where the Purging volume setting is used)

marcottt, 2019-11-04:
Hi @lukasmatena , thanks for answers. Just find something (and i think advanced option on tower purge is more simple than the "simple option").  Just another question about, the ramming is the action forced on the tower bridge just finished print old  material and before the real parking fase ?

lukasmatena, 2019-11-04:
I'm not sure I understand. Anyway, anything extruded on the tower before the toolchange with the old material is what we call ramming. It can be customized (and even turned off) in the config I pointed you to.

marcottt, 2019-11-05:
perfect. thanks

closed 2019-11-05 by lukasmatena