prusa3d/PrusaSlicer #3134: System requirements?

kssi, 2019-11-04:
System requirements – especially supported OS versions – cannot be found anywhere.

My computer is stuck on macOS 10.11.6 El Capitan and it feels highly irritating not to be able to tell whether your software will run on it.

This is a very basic and essential piece of information that should be clearly stated.

All I found so far is "Official Mac PrusaSlicer builds are currently built against SDK 10.9 to ensure compatibility with older Macs", which is not satisfying.

bubnikv, 2019-11-04:
> All I found so far is "Official Mac PrusaSlicer builds are currently built against SDK 10.9 to ensure compatibility with older Macs", which is not satisfying.

The way I read this information is: PrusaSlicer runs on OSX 10.9 and newer.

kssi, 2019-11-04:
> The way I read this information is

Because you are a developer.

I ended up finding that cryptic bit of information in a file called _How to build - Mac OS.md_, buried deep inside a Git repository.

I'm not a developer and I bet most of your userbase isn't either. You cannot expect the average user to read and understand such technical details about a task they aren't supposed to accomplish themselves in the first place.

Instead, they will read https://www.prusa3d.com/prusaslicer/ or https://www.prusa3d.com/drivers/, and download the .dmg that was created for them.

I'm not alone being in trouble: https://forum.prusaprinters.org/forum/prusaslicer/what-is-the-system-requirement-for-mac/. This thread is still left unanswered as of today.

bubnikv, 2020-03-15:
Thanks. We are currently working on a PrusaSlicer manual, we will add system requirements there.

commit 2e2e4e68 (2016-05-03): Fix layer time slowdown

The recent GCode writer changes which put the speed changes on a line of
their own have caused the layer time slowdown to be ignored by the regex
in CoolingBuffer.pm.

Fixes: #3134
closed 2020-03-15 by bubnikv