prusa3d/PrusaSlicer #3135: Error Message from PrusaSlicer 2.1.0 win32

MaximeThibaultPOLY, 2019-11-04:
[image]
### Version
PrusaSlicer 2.1.0+win32-201909160915

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
win7 32bits (build 7601) SP1
Total Ram Size RAM:  3,752MB

### 3D printer brand / version + firmware version (if known)
I3 MK3S + 3.7.2-2363

### Behavior
Unable to perform Gcode Generation and save
This is 1 STL file from previous Slicer.
I tried 1 object and 35 instance, same beaviour. The slicing is faster the error message is the same.

* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
*Saving the file
* _Actual Results_    Error Message 
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_ NO
[Moyeu9.zip](https://github.com/prusa3d/PrusaSlicer/files/3804515/Moyeu9.zip)

This is my first soumission ever on prusa and on Github.

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

Kachidoki2807, 2019-11-04:
Hello,
Try to switch the gyroid infill to the cubic one. Gyroid is known to be a huge RAM consumer.

MaximeThibaultPOLY, 2019-11-04:
Hello,

I don't use Gyroid but i did a few test with different infill pattern like cubic, the same message appear.
I did a test with Gyroid and the slicing hang.

The error i report, the slicing is done and the software hang when i try to save the Gcode.

Thank you

Kachidoki2807, 2019-11-04:
I gave a try with the provided STL with 36 instances, slice + export. The RAM usage spike at ~1.2GB. With the gyroid infill it increase to ~1.8GB. So I don't think it is a limitation from the 32bit version, but you probably really runout of RAM on your computer.

Can you monitor the RAM usage during the process?

MaximeThibaultPOLY, 2019-11-06:
After several test and memory monitoring. The problem seem to come from the computer.
memtest if failing.
I change the memory module 4 times and i still have memtest failing.
It's look like a processor or motherboard problem.
Strangely enough the os is working fine up to the moment the memory usage is around 2G. 
I will have a new (old) computer set up in the coming days.
Thank you

bubnikv, 2020-03-11:
computer error. closing.

closed 2020-03-11 by bubnikv