prusa3d/PrusaSlicer #3138: Can't slice simple model

nunofgs, 2019-11-05:
### Version
2.1.0

### Operating system type + version
Mac OS Catalina 10.15.1

### 3D printer brand / version + firmware version (if known)
MK3S

### Behavior
Can't slice this popular sock clip model:

https://www.thingiverse.com/thing:3054502

What the model looks like (pretty simple):

[image]

Here's what I get after slicing:

[image]

The tube is just gone and only 20 layers were generated.

#### Project File (.3MF) where problem occurs
[Clips_18x.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/3806829/Clips_18x.3mf.zip)

JohnOCFII, 2019-11-05:
I can confirm that I see the same results when slicing the model.  I can also say that it appears to be fixed via the free online STL repair tool at https://service.netfabb.com/service.php

[Clips_18x_fixed.stl.zip](https://github.com/prusa3d/PrusaSlicer/files/3806907/Clips_18x_fixed.stl.zip)

nunofgs, 2019-11-05:
Interesting. PrusaSlicer doesn't report any errors for me.

JohnOCFII, 2019-11-05:
> Interesting. PrusaSlicer doesn't report any errors for me.

PrusaSlicer didn't report any errors for me either, but I've gotten in the habit of running models through NetFabb if they don't slice properly.

Jebtrix, 2019-11-05:
Slices fine with latest git master.

closed 2026-09-29 by prusaslicer-bot[bot]