prusa3d/PrusaSlicer #3139: [Contributing Question] regarding DynamicPrintConfig and .ini format changes

kumpuu, 2019-11-05:
Hello,
I'm in the process of replacing the old message dialog that pops up when you exit the app and have unsaved settings. The [new one](https://i.imgur.com/Sqi8po9.png) will inform you exactly what has changed and lets you save everything.
However, in doing so I had to figure out how the settings for each different extruder are saved. There is actually only one set of (vector) settings that they all share, and extruder number corresponds to index in these vectors. That seems like a bit of a hack to me and it needlessly complicates my code so I'd like to to change it. But I'm new to slic3r and I've never worked on a big project like this so I want to make sure that such deep changes even get accepted before doing potentially a lot of work.

I propose to add `class ConfigOptionStore : public ConfigOptionSingle<DynamicPrintConfig>` which would hold all options for an extruder in one single option.
Access would be e.g. `preset->config.option("extruder_1").option(opt_key)`
It would serialize to:
```
...
extruder_1 = {
  opt_key = value
  ...
}
...
```
Which I think requires adding a file format version number to the .ini so that the app knows to translate the old extruder format. How were format changes handled in the past?

Please let me know what you think.

lukasmatena, 2019-11-06:
> it needlessly complicates my code

I don't think such a big change can be justified by this. I'm not the one to decide, but if I was, I would tell you not to do it. Ensuring backward compatibility would require other hacks and needlessly complicate the code. I assume it would be easier to just make your code work with what's there.

As for the whole extended quit dialog, it might be better to first discuss the idea with @YuSanka or @bubnikv. I'm not sure what they're up to.

kumpuu, 2019-11-06:
Ok thanks for the feedback, yesterday I went ahead without the modifications I proposed and it wasn't as bad as I thought. So problem solved.
As for discussing with @YuSanka and @bubnikv , is there an easy way to get in touch? Do you think they will read this?

closed 2019-11-12 by kumpuu